dusty-nv/jetson-inference #799: failed to load ONNX model by detectnet-camera.py (ERROR: builtin_op_importers.cpp:1554 In function importIf)

Parham-khj, 2020-11-17:
I'm trying to run detectnet-camera.py with my custom model ( trained with Object detection API 2 and converted to ONNX format with tensorflow-onnx ). With this command line  (python detectnet-camera.py --model=~/SSD_modifed_model_TRT.onnx --class_labels=~/Meat_label.txt) I've got the error below . I was wondering if could find a solution for this.

ERROR: builtin_op_importers.cpp:1554 In function importIf:
[8] Assertion failed: cond.is_weights() && cond.weights().count() == 1 && "If condition must be a initializer!"
[TRT]    failed to parse ONNX model '/home/bl/Desktop/Workplace/projects/Meat_stag1/ssd_mobilenet_V2/ONNX_model/SSD_modifed_model_TRT.onnx'
[TRT]    device GPU, failed to load /home/bl/Desktop/Workplace/projects/Meat_stag1/ssd_mobilenet_V2/ONNX_model/SSD_modifed_model_TRT.onnx
[TRT]    detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load network
Traceback (most recent call last):
  File "detectnet-camera.py", line 49, in <module>
    net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)
Exception: jetson.inference -- detectNet failed to load network

dusty-nv, 2020-11-24:
Hi @Parham-khj, it appears that TensorRT is not able to load this ONNX file from TensorFlow.

Here is another project that converts TensorFlow SSD-Mobilenet-v2 model to TensorRT:

https://github.com/AastaNV/TRT_object_detection

Or you might wish to try TF-TRT instead if you encounter issues.

mplaxman, 2021-12-01:
[image]

hi we are trying to solve above error but not able to do.few members are saying it is version problem if it is version problem how to change version please help me anyone

dusty-nv, 2021-12-01:
Hi @mplaxman, where did the model originally come from?   The SSD-Mobilenet ONNX models that are known to work were trained with the `train_ssd.py` script from this repo.

mplaxman, 2021-12-05:
hi i am using this link for github repo https://github.com/winter2897/Real-time-Auto-License-Plate-Recognition-with-Jetson-Nano

i am using model from this repo may be this model with version we are using different version am i correct ? if i correct i am not able to update tensorrt

dusty-nv, 2021-12-06:
I'm not familiar with how they trained those models - what I do know is that SSD-Mobilenet ONNX works with TensorRT when trained with PyTorch and the `train_ssd.py` script from the repo.   It does look like they used Google Colab and have some notebooks for the training.

Perhaps they were using a different version of JetPack?   You may want to file a question against their GitHub for more info.

mplaxman, 2021-12-07:
using this script train_ssd.py i done training but not able to convert onnx format it is giving size mismatch error any idea about this ? and is it possible to inference without onnx in jetson inference ?

dusty-nv, 2021-12-07:
Unfortunately this model seems incompatible with the version of TensorRT that you are running.  What version of JetPack-L4T are you running?  (you can check this with `cat /etc/nv_tegra_release`)

How did you train the model?

mplaxman, 2021-12-09:
[image]

i trained using train_ssd.py

Puliesi, 2022-09-13:
Estou com o mesmo problema:

ERROR: ModelImporter.cpp:296 In function importModel:
[5] Assertion failed: tensors.count(input_name)
[TRT]    failed to parse ONNX model 'models/humanos/ssd-mobilenet.onnx'
[TRT]    device GPU, failed to load models/humanos/ssd-mobilenet.onnx
[TRT]    detectNet -- failed to initialize.
detectnet:  failed to load detectNet model
marcos@marcos-desktop:~/jetson-inference/python/training/detection/ssd$ 

Não consigo carregar o modelo usando o DetectNet.
Minha configuração:

NVIDIA NVIDIA Jetson Nano Developer Kit
 L4T 32.4.4 [ JetPack 4.4.1 ]
   Ubuntu 18.04.6 LTS
   Kernel Version: 4.9.140-tegra
 CUDA 10.0.326
   CUDA Architecture: 5.3
 OpenCV version: 4.1.1
   OpenCV Cuda: NO
 CUDNN: 7.6.3.28
 TensorRT: 7.1.3.0
 Vision Works: 1.6.0.501
 VPI: ii vpi 0.4.4 arm64 Vision Programming Interface
 Vulcan: 1.2.70

Por favor, alguem pode me ajudar?
Agradeço antecipadamente!

closed 2023-04-27 by dusty-nv