dusty-nv/jetson-inference #800: Custom ONNX model load problem

choi0330, 2020-11-17:
Hello dusty! I am very much enjoying your tutorials and grateful for it.

I've looked through the issues but I couldn't find a relevant issue.

I am currently trying to load .onnx model (which is not from one of your tutorials) with detectNet on my custom C++ code. (jetson-inference is correctly included)
Using C++ detecNet class, `net = detectNet::Create(NULL, "path/to/model.onnx", "path/to/labels.txt");`

The error occurs when the internal code is trying to read input_blobs.
```
[TRT]    INVALID_ARGUMENT: Cannot find binding of given name: input_0
[TRT]    failed to find requested input layer input_0 in network
[TRT]    device GPU, failed to create resources for CUDA engine
````

I noticed that the default argument for `input_blob` doesn't work for external .onnx model.

Should I provide a correct input_blob argument to load the model? And how can I know this information?
Most of the conversion examples (anymodel to .onnx) don't provide this information, so I need some help on this with jetson-inference.

Looking forward to your ideas! Thanks :)

murthax, 2020-11-18:
I'm having the same problem with a model I converted from PyTorch Yolov5. I get:

[TRT]    INVALID_ARGUMENT: Cannot find binding of given name: data
[TRT]    failed to find requested input layer data in network
[TRT]    device GPU, failed to create resources for CUDA engine
[TRT]    failed to create TensorRT engine for models/test/best.onnx, device GPU
[TRT]    detectNet -- failed to initialize.
detectnet:  failed to load detectNet model

I tried a few other options I had from other models like --input-blob=input_0 but without knowing where to look for this I wasn't sure.

I am using the aarch64 Jetson Xavier NX.

choi0330, 2020-11-18:
@murthax I am trying the same model Pytorch Yolov5. Also, my platform is aarch64 Jetson Xavier AGX.

And I have found the input blob names and output names.
Could you try --input-blob=images?

However, when you export the model to onnx with export.py (from yolov5 source code), you will notice that there is only one output name which is 'output' since dry run will work from [here](https://github.com/ultralytics/yolov5/blob/master/models/export.py#L69)

Then, I assign them as output_names=['classes', 'boxes']. This way jetson detectNet can load the model but I have further issues.

My current issue is that output binding doesn't seem to be correct which doesn't complain in the loading stage. It looks like following:
```
[TRT]    binding to input 0 images  binding index:  0
[TRT]    binding to input 0 images  dims (b=1 c=3 h=736 w=544) size=4804608
[TRT]    binding to output 0 classes  binding index:  1
[TRT]    binding to output 0 classes  dims (b=1 c=3 h=92 w=68) size=75072
[TRT]    binding to output 1 boxes  binding index:  2
[TRT]    binding to output 1 boxes  dims (b=1 c=3 h=46 w=34) size=18768
```
(I tweaked the input image size as well when I export the model.)

Then, when I try to run detection with this model, I get an illegal memory access error in CUDA.
```
detection start
[TRT]    engine.cpp (986) - Cuda Error in executeInternal: 700 (an illegal memory access was encountered)
[TRT]    FAILED_EXECUTION: std::exception
[TRT]    failed to execute TensorRT context on device GPU
detection end
```
The first and last print are my debugging messages.

berkantay, 2020-11-18:
I am having the same issue for jetsontx2 with YOLOv5. I get exactly same error. Any update would be helpful.

choi0330, 2020-11-18:
@berkantay Could you check your model with [neutron](https://netron.app/)?

It seems like detection stage is not included in the onnx model conversion. [This](https://github.com/ultralytics/yolov5/issues/343) might help..

I am encountering two more issues now.

1. ONNX model loading outputs plugin issue when I include the detection stage in the model. (putting

https://github.com/ultralytics/yolov5/blob/a1c8406af3eac3e20d4dd5d327fd6cbd4fbb9752/models/export.py#L28

 into **False**.)
```
[TRT]    builtin_op_importers.cpp:3659: Searching for plugin: ScatterND, plugin_version: 1, plugin_namespace: 
[TRT]    INVALID_ARGUMENT: getPluginCreator could not find plugin ScatterND version 1
ERROR: builtin_op_importers.cpp:3661 In function importFallbackPluginImporter:
[8] Assertion failed: creator && "Plugin not found, are the plugin name, version, and namespace correct?"
```

2. How can we make an output corresponding to the jetson detectNet? (including NMS)

berkantay, 2020-11-18:
> @berkantay Could you check your model with [neutron](https://netron.app/)?
> 
> It seems like detection stage is not included in the onnx model conversion. [This](https://github.com/ultralytics/yolov5/issues/343) might help..
> 
> I am encountering two more issues now.
> 
>     1. ONNX model loading outputs plugin issue when I include the detection stage in the model. (putting
> 
> 
> https://github.com/ultralytics/yolov5/blob/a1c8406af3eac3e20d4dd5d327fd6cbd4fbb9752/models/export.py#L28
> 
> into **False**.)
> 
> ```
> [TRT]    builtin_op_importers.cpp:3659: Searching for plugin: ScatterND, plugin_version: 1, plugin_namespace: 
> [TRT]    INVALID_ARGUMENT: getPluginCreator could not find plugin ScatterND version 1
> ERROR: builtin_op_importers.cpp:3661 In function importFallbackPluginImporter:
> [8] Assertion failed: creator && "Plugin not found, are the plugin name, version, and namespace correct?"
> ```
> 
>     1. How can we make an output corresponding to the jetson detectNet? (including NMS)

Hello @choi0330 when I check the model with neutron input layer is called *images*. And this is the output of the converted model. https://ibb.co/TTSQw5r

murthax, 2020-11-18:
I'm at a similar stage. I did use Netron to interpret the ONNX file. I see some outputs. I was able to move past the input-blob issue by using "images" but then on to the next ones (--input-blob=images --output-cvg=? --output-bbox=?

[image]

If I use the outputs here ie. --input-blob=images --output-cvg=791 --output-bbox=771 the engine starts but I end up with memory errors:

[TRT]    engine.cpp (986) - Cuda Error in executeInternal: 700 (an illegal memory access was encountered)
[TRT]    FAILED_EXECUTION: std::exception
[TRT]    failed to execute TensorRT context on device GPU
[OpenGL] glDisplay -- set the window size to 1280x720
[OpenGL] creating 1280x720 texture (GL_RGB8 format, 2764800 bytes)
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/magneto/jetson-inference/utils/display/glTexture.cpp:360
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/magneto/jetson-inference/build/aarch64/include/jetson-utils/cudaMappedMemory.h:51
RingBuffer -- failed to allocate zero-copy buffer of 1382400 bytes
[gstreamer] gstEncoder -- failed to allocate buffers (1382400 bytes each)
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/magneto/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685

dusty-nv, 2020-11-18:
To support different object detection models in jetson-inference, you would need to add/modify the pre/post-processing code found here:

* https://github.com/dusty-nv/jetson-inference/blob/c8a0392b1b49d9892aa4ec0556eb058f5d2e7145/c/detectNet.cpp#L728
* https://github.com/dusty-nv/jetson-inference/blob/c8a0392b1b49d9892aa4ec0556eb058f5d2e7145/c/detectNet.cpp#L815

This should be made to match the pre/post-processing that gets performed on the original model.
It also seems like you might need to add support for a 3rd output layer - the previous detection models in jetson-inference used 2 output layers.

Since you are using PyTorch, you might also want to try the torch2trt project - https://github.com/nvidia-ai-iot/torch2trt

choi0330, 2020-11-19:
Thanks for the answer.

I would love to utilize Jentson-inference library, so I'll try to match the pre/post-processing of the original model for now.

berkantay, 2020-11-19:
If you find out the solution it would be nice for us @choi0330. I will be also updating this issue in case of good updates.

berkantay, 2020-11-20:
Hello everyone when I change the command with ```detectnet --model=/home/user/Downloads/tyes.onnx --labels=/home/dasal/Downloads/a.txt --input-blob=input_0 --output-cvg=771 --output-bbox=791 
```

I see no memory warnings network works. But I get errors from gstreamer unsupported image format.

```detectnet:  failed to capture video frame
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/dasal/Projects/jetson-inference/utils/cuda/cudaYUV-NV12.cu:154
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/dasal/Projects/jetson-inference/utils/cuda/cudaColorspace.cpp:42
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/dasal/Projects/jetson-inference/utils/codec/gstDecoder.cpp:895
[gstreamer] gstDecoder::Capture() -- unsupported image format (rgb8)
[gstreamer]                          supported formats are:
[gstreamer]                              * rgb8
[gstreamer]                              * rgba8
[gstreamer]                              * rgb32f
[gstreamer]                    
```

choi0330, 2020-11-20:
Here I found some hint on how to export the model with the detection layer: https://github.com/ultralytics/yolov5/issues/708#issuecomment-674422178

However, the default TensorRT doesn't support a Scatter ND operation which is in the detection layer. So, one possible option would be to implement the post-processing using three outputs (https://github.com/ultralytics/yolov5/blob/master/models/yolo.py#L52-L59) and NMS.

@murthax I share my current update. 
@berkantay Please make sure your output layer `--output-cvg=771 --output-bbox=791` are confidences and bounding boxes.

RGring, 2021-01-20:
Hi @choi0330 @berkantay,
I am also interested in using ultralytics yolov5 models within the jetson-inference framework. Did you get it working? I would appreciate your help!

murthax, 2021-01-20:
> Hi @choi0330 @berkantay,
> I am also interested in using ultralytics yolov5 models within the jetson-inference framework. Did you get it working? I would appreciate your help!

I wish I had better news but I was not able to move forward here. I've been using Yolov5 directly on my Jetson Xavier NX. The FPS isn't as good but at least I'm able to move forward with my project.

berkantay, 2021-01-21:
> Hi @choi0330 @berkantay,
> I am also interested in using ultralytics yolov5 models within the jetson-inference framework. Did you get it working? I would appreciate your help!

Unfortunately I stopped my work on jetson-inference and implemented [yolov5 ](https://github.com/ultralytics/yolov5)itself without jetson-inference library on jetson tx2.

RGring, 2021-01-22:
Ok thanks for your responses!

choi0330, 2021-01-22:
> Hi @choi0330 @berkantay,
> I am also interested in using ultralytics yolov5 models within the jetson-inference framework. Did you get it working? I would appreciate your help!

I went a bit further to implement detection layers in Jetson-inference library but it didn't work out. I guess it will be best to wait for the plugin of Scatter ND operation in TensorRT to run yolov5 with Jetson-inference.
In the end, I also run Yolov5 directly on Jetson for now.

dusty-nv, 2021-01-22:
It looks like there are some TensorRT YOLOv5 projects out there:

https://github.com/SeanAvery/yolov5-tensorrt
https://www.google.com/search?q=tensorRT+yolov5

MichaelWU0726, 2021-08-17:
> Here I found some hint on how to export the model with the detection layer: [ultralytics/yolov5#708 (comment)](https://github.com/ultralytics/yolov5/issues/708#issuecomment-674422178)
> 
> However, the default TensorRT doesn't support a Scatter ND operation which is in the detection layer. So, one possible option would be to implement the post-processing using three outputs (https://github.com/ultralytics/yolov5/blob/master/models/yolo.py#L52-L59) and NMS.
> 
> @murthax I share my current update.
> @berkantay Please make sure your output layer `--output-cvg=771 --output-bbox=791` are confidences and bounding boxes.

Operator `scatterND` has been supported in version 8, could you migrate the implementation of the operator to version 7 ? I have to use version 7.

closed 2023-04-27 by dusty-nv