dusty-nv/jetson-inference #801: Re-training SSD-Mobilenet

AhmetEnesYalcinkaya, 2020-11-17:
Hello, I have a problem about Re-training SSD-Mobilenet in fruit example..

I trained my model and I created ssd-mobilenet.onnx file. you look below.. But I take this error. I checked my folder path. ıt seen correct. Please Can you help me ?

[image]

dusty-nv, 2020-11-17:
Hi @AhmetEnesYalcinkaya , it is finding your model, but failing to parse it correctly (with the `tensors.count(input_name)` error)

Which version of JetPack are you running?

Can you try using the jetson-inference container to see if that works for you?   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md

AhmetEnesYalcinkaya, 2020-11-24:
Thank you for solve. It worked fine..

I  wonder it "How I can change screen size (I want to do fullscreen) ?" and "How I can apply seen just box. don't conf etc.?( I used --overlay but it didnt worked")
 
Thank you again.

dusty-nv, 2020-11-24:
> "How I can apply seen just box. don't conf etc.?( I used --overlay but it didnt worked")

You want to launch detectnet program with the `--overlay=box` option, for example:

``` bash
$ detectnet --overlay=box /dev/video0
```

> "How I can change screen size (I want to do fullscreen) ?"

You need to make a couple edits to [`jetson-inference/examples/detectnet/detectnet.cpp`](https://github.com/dusty-nv/jetson-inference/blob/master/examples/detectnet/detectnet.cpp):

1. Add an `#include "glDisplay.h"` near the top
2. Add these lines below [line 158](https://github.com/dusty-nv/jetson-inference/blob/8498115a4ed8967c7e23f5ae27b5aa2fa1d1431b/examples/detectnet/detectnet.cpp#L158):

``` cpp
if( output->IsType<glDisplay>() )
     ((glDisplay*)output)->SetMaximized(true);
```

3.  `cd jetson-inference/build` and re-run `make` and `sudo make install`

AhmetEnesYalcinkaya, 2020-11-25:
It worked fine. (detectnet --overlay=box /dev/video0)
- But I am wondering it why didnt it work in my-detection.py?

For example : net = jetson.inference.detectNet(argv=['--model=models/ssd-mobilenet.onnx', '--labels=models/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes', **'--overlay=box,label'**], threshold=0.5)

- Can I do fullsize in my_detection.py ?

dusty-nv, 2020-11-25:
> * But I am wondering it why didnt it work in my-detection.py?
> 
> For example : net = jetson.inference.detectNet(argv=['--model=models/ssd-mobilenet.onnx', '--labels=models/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes', **'--overlay=box,label'**], threshold=0.5)

That's because the overlay flags are an (optional) argument to `detectNet.Detect()` as opposed to the constructor.

You can add this to your `net.Detect()` call in `my-detection.py`:

``` python
detections = net.Detect(img, overlay='box')
```

> * Can I do fullsize in my_detection.py ?

I think you would need to edit the glDisplay C++ implementation in jetson-utils.   Comment out this code block from your [`jetson-inference/utils/display/glDisplay.cpp:649`](https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/display/glDisplay.cpp#L649) :

``` cpp
              /*if( !mResizedToFeed || ((GetWidth() < width || GetHeight() < height) && (width < mScreenWidth && height < mScreenHeight)) )
		{
			SetSize(width, height);
			mResizedToFeed = true;
		}*/
```

Then re-compile by running `make` and `sudo make install` again.

AhmetEnesYalcinkaya, 2020-12-01:
> ```c++
> if( output->IsType<glDisplay>() )
>      ((glDisplay*)output)->SetMaximized(true);
> ```
((glDisplay*)output)   give errorr ! you can check it this line ? @dusty-nv

dusty-nv, 2020-12-01:
What was the error?  Did you also #include "glDisplay.h"?