dusty-nv/jetson-inference #802: Is it possible to use desktop GPU to train?

ysyyork, 2020-11-17:
I saw all the tutorial on training is for pytorch on jetson. Is it possible to use pytorch on a desktop server with desktop GPUs like 2080, 3080 etc and then export onto jetson?

dusty-nv, 2020-11-17:
Hi @ysyyork , yes, I commonly do this here when training on larger datasets.  The PyTorch training code is organized into git submodules so that you can clone just those modules on your PC - for example:

* classification - https://github.com/dusty-nv/pytorch-classification
* detection - https://github.com/dusty-nv/pytorch-ssd

The commands you run on your PC would essentially be the same as you run in the Hello AI World tutorial.  You can probably use a larger batch size though.  After you export the model to ONNX, just copy that ONNX model over to your Jetson.

My PC has Ubuntu on it, and I installed PyTorch according to the instructions from pytorch.org.

ysyyork, 2020-11-18:
Thanks! this helps a lot!

So this also means any pytorch model that can be converted to ONNX should
be able to be run on jetson right? doesn't need to be the models in this
repo?

dusty-nv, 2020-11-18:
> So this also means any pytorch model that can be converted to ONNX should be able to be run on jetson right? doesn't need to be the models in this repo?

You would need to make sure that TensorRT can load the ONNX model (you can test this with the trtexec tool found under `/usr/src/tensorrt/bin` on your Jetson).  Also if you actually wanted to run the ONNX model on real data, you may need to add the pre/post-processing code into jetson-inference - these are typically equivalent to the transforms that torchvision does on the input tensor(s) before the model is run.

Another option for running custom PyTorch models with TensorRT is to use the torch2trt project - https://github.com/NVIDIA-AI-IOT/torch2trt
It transparently accelerates PyTorch models with TensorRT while keeping the same PyTorch API