dusty-nv/jetson-inference #804: How to upgrade the jetson-inference?

AK51, 2020-11-18:
I installed the jetson-inference in Jan 2019 and it is great. I have created built mysql, node js around it.
And I saw there is a new version of jetson-inference with custom object detection. I assume I can rename the old jetson-inference folder to something else, and install the new jetson-inference version, right?

Thanks.

dusty-nv, 2020-11-18:
Hi @AK51, yes you can do that, or you could also clone the new copy of jetson-inference to a different directory:

``` bash
$ git clone --recursive https://github.com/dusty-nv/jetson-inference jetson-inference-new
```

Note that when you do the `sudo make install` step, the new jetson-inference will be installed under `/usr`.  But you can re-install the old one by simply running `sudo make install` again in the old one's build dir.

AK51, 2020-11-21:
Thx.
I end up making a new SD card and install the mySQL and node.js.
Never mind, it is working now.
Thanks,
Andy

closed 2020-11-21 by AK51