dusty-nv/jetson-inference #805: Run two demo using one webcam?

AK51, 2020-11-18:
Hi, I can get the live demo working, and I want to run two applications (different custom model, one classification and one object detection) using v4l2loopback.
The webcam, dev/video0, works, but the dummy stream of dev/video1 using v4l2loopback stuck at gststreamer
I wonder if anyone has tried the v4l2loopback on the demo before?

Or may be I run two models in a single python program.

Thx

dusty-nv, 2020-11-18:
Hi @AK51, it would be more efficient (and perhaps easier for you to manage) to run both models in the same program.

You would want to 'hardcode' your custom model options as opposed to parsing them from the command line, because the command line parsing done by imageNet / detectNet isn't aware of multiple models.  

So instead, do something like below  (this example is for custom detectNet model)

``` python
detectionNet = jetson.inference.detectNet(argv=['--model=models/ssd-mobilenet.onnx', '--labels=models/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes'], threshold=0.5)
```

AK51, 2020-11-22:
Yep, it works. Thx

closed 2020-11-22 by AK51