dusty-nv/jetson-inference #806: Error during model training ( OSError: [Errno 12] Cannot allocate memory)

michael-herm, 2020-11-18:
I created a detection dataset and tried to run the model training. then this error occured: OSError: [Errno 12] Cannot allocate memory
what went wrong?

root@michael-desktop:/jetson-inference/python/training/detection/ssd# python3 train_ssd.py --dataset-type=voc --data=data/sweets3 --model-dir=models/sweets3 --batch-size=2 --workers=1 --epochs=1
2020-11-18 12:29:40 - Using CUDA...
2020-11-18 12:29:40 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=2, checkpoint_folder='models/sweets3', dataset_type='voc', datasets=['data/sweets3'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=1, num_workers=1, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-11-18 12:29:40 - Prepare training datasets.
2020-11-18 12:29:40 - VOC Labels read from file: ('BACKGROUND', 'airwaves', 'icebreakers', '')
2020-11-18 12:29:40 - Stored labels into file models/sweets3/labels.txt.
2020-11-18 12:29:40 - Train dataset size: 182
2020-11-18 12:29:40 - Prepare Validation datasets.
2020-11-18 12:29:40 - VOC Labels read from file: ('BACKGROUND', 'airwaves', 'icebreakers', '')
2020-11-18 12:29:40 - Validation dataset size: 20
2020-11-18 12:29:40 - Build network.
2020-11-18 12:29:40 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2020-11-18 12:29:41 - Took 0.50 seconds to load the model.
2020-11-18 12:29:59 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2020-11-18 12:29:59 - Uses CosineAnnealingLR scheduler.
2020-11-18 12:29:59 - Start training from epoch 0.
/usr/local/lib/python3.6/dist-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/usr/local/lib/python3.6/dist-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
2020-11-18 12:36:48 - Epoch: 0, Step: 10/91, Avg Loss: 11.2521, Avg Regression Loss 4.2961, Avg Classification Loss: 6.9560
2020-11-18 12:37:17 - Epoch: 0, Step: 20/91, Avg Loss: 9.4423, Avg Regression Loss 4.4821, Avg Classification Loss: 4.9603
2020-11-18 12:48:39 - Epoch: 0, Step: 30/91, Avg Loss: 6.0758, Avg Regression Loss 2.4913, Avg Classification Loss: 3.5845
2020-11-18 12:49:28 - Epoch: 0, Step: 40/91, Avg Loss: 5.7719, Avg Regression Loss 2.3850, Avg Classification Loss: 3.3869
2020-11-18 12:49:33 - Epoch: 0, Step: 50/91, Avg Loss: 5.9104, Avg Regression Loss 2.4241, Avg Classification Loss: 3.4863
2020-11-18 12:49:48 - Epoch: 0, Step: 60/91, Avg Loss: 4.7955, Avg Regression Loss 2.0276, Avg Classification Loss: 2.7678
2020-11-18 12:50:34 - Epoch: 0, Step: 70/91, Avg Loss: 5.3096, Avg Regression Loss 2.2703, Avg Classification Loss: 3.0393
2020-11-18 12:50:48 - Epoch: 0, Step: 80/91, Avg Loss: 4.8920, Avg Regression Loss 1.9360, Avg Classification Loss: 2.9560
2020-11-18 12:51:30 - Epoch: 0, Step: 90/91, Avg Loss: 4.7582, Avg Regression Loss 1.8124, Avg Classification Loss: 2.9458
Traceback (most recent call last):
  File "train_ssd.py", line 346, in <module>
    val_loss, val_regression_loss, val_classification_loss = test(val_loader, net, criterion, DEVICE)
  File "train_ssd.py", line 150, in test
    for _, data in enumerate(loader):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 282, in __iter__
    return _MultiProcessingDataLoaderIter(self)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 728, in __init__
    w.start()
  File "/usr/lib/python3.6/multiprocessing/process.py", line 105, in start
    self._popen = self._Popen(self)
  File "/usr/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/usr/lib/python3.6/multiprocessing/context.py", line 277, in _Popen
    return Popen(process_obj)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 66, in _launch
    self.pid = os.fork()
OSError: [Errno 12] Cannot allocate memory
Segmentation fault (core dumped)
root@michael-desktop:/jetson-inference/python/training/detection/ssd# ^C
root@michael-desktop:/jetson-inference/python/training/detection/ssd#

dusty-nv, 2020-11-18:
My guess is that the system ran out of memory.  Do you have SWAP memory mounted?  Can you run `sudo tegrastats` while running the training to keep an eye on the memory consumption?

You can also try disabling the Ubuntu GUI while training, this will save you extra memory:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

michael-herm, 2020-11-18:
You were right. It was the missig SWAP memory.
Thanks for the quick response!

closed 2020-11-18 by michael-herm

dusty-nv, 2022-09-21:
Hi @Abhishek183422, see these steps to mount swap:   https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-995090398