dusty-nv/jetson-inference #809: use gpu to decode and my cpu is high high high

ruiming46zrm, 2020-11-20:
codes as follwing,    my disire is to use gpu decode and release my cpu , but which makes my cpu high hight high, i guess it is because the mencopy takes, but how can I deal with it?  I want to do some work in cpu.
```
import jetson.utils
import cv2

camera = jetson.utils.videoSource("rtsp://admin:lettin20@192.168.2.220:554/Streaming/Channels/1")      # '/dev/video0' for V4L2
while True:
    img_o = camera.Capture()
    jetson.utils.cudaDeviceSynchronize()
    img = jetson.utils.cudaToNumpy(img_o)
    cv2.imshow('a', img)
    cv2.waitKey(1)`
```

dusty-nv, 2020-11-20:
Is the CPU usage reduced if you remove these lines?

``` bash
cv2.imshow('a', img)
cv2.waitKey(1)
```

cudaToNumpy() shouldn't make an extra copy, it maps the image into numpy array.

ruiming46zrm, 2021-01-22:
hello, here I am  again,   this is my time spend:
```
decode time: 2.8848648071289062e-05
cudaAllocateMapped time: 0.04077744483947754
cuda cvt color time: 9.298324584960938e-05
cuda to numpy: 0.039249420166015625
show time: 0.07740068435668945
```

I am wondering about the cuda to numpy time and cudaAllocateMapped time .  it's quite a lot.
really thank you for your reply to beginners like me ~