dusty-nv/jetson-inference #810: no element "omxh264dec"

SergeyMalashenko, 2020-11-20:
Good day! I am trying to build and use your project. I am getting the following message.

[gstreamer] gstDecoder -- failed to create pipeline
[gstreamer]    (no element "omxh264dec")
[gstreamer] gstDecoder -- failed to create decoder for file:///home/smalashenko/My_projects/C++/jetson-inference/build/x86_64/bin/./e36-test-29294ba2ae_1602173004_587.mp4
detectnet:  failed to create input stream

dusty-nv, 2020-11-20:
Hi @SergeyMalashenko , it appears you are building this on x86_64, so the jetson-utils video decoding won't work there (as those hardware-accelerated gstreamer elements are for Jetson only)

You can try changing the GStreamer element here:

https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/codec/gstDecoder.cpp#L554

You would want to change it from `omxh264dec` to a h264 decoding plugin for GStreamer that you have on x86 (see [here](https://stackoverflow.com/a/43466180))

irshadkn, 2021-07-14:
**Hardware: Jetson Nano 4GB**

video-viewer --input-codec=h265 rtsp://admin:rgbsi123@192.168.1.61:554/Streaming/channels/101
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin:rgbsi123@192.168.1.61
Opening in BLOCKING MODE
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 279 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 279 
[gstreamer] gstDecoder -- discovered video resolution: 1280x720  (framerate 0.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1280, height=(int)720, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)main, tier=(string)main, level=(string)3.1
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://@192.168.1.61:554/Streaming/channels/101 latency=2000 ! queue ! rtph265depay ! h265parse ! omxh265dec ! video/x-raw ! appsink name=mysink
[gstreamer] gstDecoder -- failed to create pipeline
**[gstreamer]    (no element "omxh265dec")**
[gstreamer] gstDecoder -- failed to create decoder for rtsp://@192.168.1.61:554/Streaming/channels/101
video-viewer:  failed to create input stream

omxh264dec and omxh265dec not found on jetson nano

dusty-nv, 2021-07-14:
Can you run `gst-inspect-1.0 | grep omx` ?   Are there any listed?

These should be automatically installed when you use the Nano SD card image.

irshadkn, 2021-07-14:
**gst-inspect-1.0 | grep omx**
libav:  avenc_h264_omx: libav OpenMAX IL H.264 video encoder encoder

I downloaded the image from this link:
https://developer.nvidia.com/jetson-nano-sd-card-image

dusty-nv, 2021-07-14:
This is the expected output:

``` bash
gst-inspect-1.0 | grep omx

libav:  avenc_h264_omx: libav OpenMAX IL H.264 video encoder encoder
omx:  omxmpeg4videodec: OpenMAX MPEG4 Video Decoder
omx:  omxh264dec: OpenMAX H.264 Video Decoder
omx:  omxh265dec: OpenMAX H.265 Video Decoder
omx:  omxvp8dec: OpenMAX VP8 Video Decoder
omx:  omxvp9dec: OpenMAX VP9 Video Decoder
omx:  omxmpeg2videodec: OpenMAX MPEG2 Video Decoder
omx:  omxwmvdec: OpenMAX WMV Video Decoder
omx:  omxh264enc: OpenMAX H.264 Video Encoder
omx:  omxh265enc: OpenMAX H.265 Video Encoder
omx:  omxvp8enc: OpenMAX VP8 Video Encoder
omx:  omxvp9enc: OpenMAX VP9 Video Encoder
omx:  nvoverlaysink: OpenMax Video Sink
```

You may want to try re-flashing your SD card and confirm what this outputs with a fresh image.