dusty-nv/jetson-inference #813: Error:No module named 'jetson'

wuzhishiwo, 2020-11-22:
I have not started the container,I don't know if this step is necessary, I just run the .py and the image, and then the question occured, if you know how to solve,please tell me.

wuzhishiwo, 2020-11-23:
-

wuzhishiwo, 2020-11-23:
[image]

wuzhishiwo, 2020-11-23:
[image]

dusty-nv, 2020-11-24:
Hi @wuzhishiwo , if you did not [Build from Source](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md), then yes you would need to [Run the Container](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md) first.

When starting the container, you can mount your `my-recognition-python` directory into the container like so:

``` bash
docker/run.sh --volume ~/my-recognition-python:/my-recognition-python
```
It will then show up inside the container under `/my-recognition-python`

closed 2023-04-27 by dusty-nv