dusty-nv/jetson-inference #814: MobileNet-v1 transfer learning LOSS = INF at every save.

pradhunmya01, 2020-11-23:
Hi @dusty-nv I am also getting the issue but here it comes while saving the model it saves like:-
mb1-ssd-Epoch-0-Loss-inf.pth
mb1-ssd-Epoch-1-Loss-inf.pth
mb1-ssd-Epoch-2-Loss-inf.pth
[image]

dusty-nv, 2020-11-24:
Hmm I wonder if there is some corrupt or incorrect annotations in your dataset?

You may need to drill down and determine which image(s) are throwing it off.   Here are some techniques which can help you narrow it down:

* set `shuffle=False`  in [train_ssd.py:237](https://github.com/dusty-nv/pytorch-ssd/blob/8070c17770d4ebef1b3a4092722168c701fc0a0c/train_ssd.py#L237)
* uncomment this debug `print()` in [voc_dataset.py:76](https://github.com/dusty-nv/pytorch-ssd/blob/8070c17770d4ebef1b3a4092722168c701fc0a0c/vision/datasets/voc_dataset.py#L76)
* run the program with `--debug-steps=1` option

Setting `--debug-steps=1` should let you see which batch is causing the INF overflow.  Then that print statement will let you know what images were loaded for that batch to check.   Setting `shuffle=False` will make it run in the same order each time.

pradhunmya01, 2020-11-25:
Thank you @dusty-nv for the great suggestion, I tried the same process and get the annotation file but I doesn't seem any problem with annotation file and image please check:-

This is that annotation file and image is also perfect.
<annotation>
	<folder>input-cropped-shelf-images-wba-03406-c003</folder>
	<filename>1470-wba03406000c003-1600206461263-middle-shelf-1.jpg</filename>
	<path>D:\Megha\Object Tagging\_Megha\input-cropped-shelf-images-wba-03406-c003\1470-wba03406000c003-1600206461263-middle-shelf-1.jpg</path>
	<source>
		<database>Unknown</database>
	</source>
	<size>
		<width>900</width>
		<height>400</height>
		<depth>3</depth>
	</size>
	<segmented>0</segmented>
	<object>
		<name>bottle</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>23</xmin>
			<ymin>148</ymin>
			<xmax>146</xmax>
			<ymax>380</ymax>
		</bndbox>
	</object>
	<object>
		<name>bottle</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>173</xmin>
			<ymin>148</ymin>
			<xmax>296</xmax>
			<ymax>377</ymax>
		</bndbox>
	</object>
	<object>
		<name>bottle</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>300</xmin>
			<ymin>144</ymin>
			<xmax>412</xmax>
			<ymax>379</ymax>
		</bndbox>
	</object>
	<object>
		<name>bottle</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>412</xmin>
			<ymin>148</ymin>
			<xmax>523</xmax>
			<ymax>379</ymax>
		</bndbox>
	</object>
	<object>
		<name>bottle</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>528</xmin>
			<ymin>148</ymin>
			<xmax>641</xmax>
			<ymax>371</ymax>
		</bndbox>
	</object>
	<object>
		<name>bottle</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>671</xmin>
			<ymin>150</ymin>
			<xmax>780</xmax>
			<ymax>364</ymax>
		</bndbox>
	</object>
	<object>
		<name>bottle</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>788</xmin>
			<ymin>150</ymin>
			<xmax>895</xmax>
			<ymax>370</ymax>
		</bndbox>
	</object>
</annotation>

dusty-nv, 2020-11-25:
There are probably other images in that batch that causes the INF overflow.  For example, if the batch size is 8, there will be 8 images that get loaded for each batch.  You could set the batch size to 2 to narrow it down to 2 images.

Unknown

900
400

bottle
Unspecified

23
148
146
380

bottle
Unspecified

173
148
296
377

bottle
Unspecified

300
144
412
379

bottle
Unspecified

412
148
523
379

bottle
Unspecified

528
148
641
371

bottle
Unspecified

671
150
780
364

bottle
Unspecified

788
150
895
370

pradhunmya01, 2020-11-25:
yes I did the batch size to 1 and check that annotation and images even its fine

pradhunmya01, 2020-11-25:
Another issue is that it showing this error but still the bndbox value is in the annotations 
[image]

dusty-nv, 2020-11-25:
> can you see anything wrong here? @dusty-nv

I don't exactly see what is wrong, but can you try removing that ImageID from the ImageSet files under `ImageSets/Main/*.txt`?   This will make it so that particular image isn't loaded.   Do you still get the error?

Which annotation tool did you use to create the dataset?

pradhunmya01, 2020-11-25:
> I don't exactly see what is wrong, but can you try removing that ImageID from the ImageSet files under `ImageSets/Main/*.txt`? This will make it so that particular image isn't loaded. Do you still [...]

I used LabelImg for annotations,
and yeah I am still getting the errors with other Images, I removed that imageID from Imageset but continuously it showing the errors on other images.

dusty-nv, 2020-11-25:
Is it the INF overflow problem you are still getting, or the `too many indices` error?    Interesting that you were not getting the later error before.

Can you send me your dataset so I can try it here?   It is holiday coming up here for the rest of this week, but I will try to look at it soon.

pradhunmya01, 2020-11-25:
Thank you so much @dusty-nv  for this much concern, actually that dataset is different for INF and this one is different but the dataset is much similar, let me share my dataset with you I am uploading it on the google drive.

dusty-nv, 2020-11-25:
OK gotcha - typically the `too many indices` error means that there is a malformed object in the XML, or no objects in the XML.   I had checked in a fix for it probably a month ago, but perhaps it didn't cover all the corner cases.

I would check the XML file that threw the error when it tried to be loaded, and make sure that it is OK.

pradhunmya01, 2020-11-25:
yeah that will be great I guess, upload is on the way.

pradhunmya01, 2020-11-25:
Hi @dusty-nv Here's the Link of dataset:- 
[Dataset Download](https://drive.google.com/file/d/16uevvFbKfXQ5Zq2xoQ7oRy61ewe7w1L7/view) 
please check and let me know if you need anything else.
Thank you in advance

pradhunmya01, 2020-11-26:
@dusty-nv I have you the access please check

dusty-nv, 2020-12-01:
Hi @pradhunmya01 , sorry for the delay while I was catching up from Thanksgiving holiday.

When I tried to run your dataset, I got a bunch of warning like this:

``` bash
warning - image 545-wba15196000c004-1600143490978-top-shelf-0 has object with unknown class 'bottlle'
warning - image 625-wba15196000c003-1600143407923-bottom-shelf-0 has object with unknown class 'bottlle'
```

These were then causing errors while trying to train the model.  In your annotations, there were some typos in the class names.  I was able to fix these by running the following commands (make a backup copy of your dataset before running these):

``` bash
cd /path/to/your/dataset/Annotations
sed -i 's/bottlle/bottle/g' *.xml
sed -i 's/bottled/bottle/g' *.xml
sed -i 's/Bottlee/bottle/g' *.xml
```

The model then is able to train.   I ran it with the default settings:

``` bash
python3 train_ssd.py --dataset-type=voc --data=$DATA --model-dir=$MODEL | tee $MODEL/train_log_20201201.txt
```

The avg loss appears high (but not INF), but the avg classification loss is ok (which is the important one to watch).   I will let it run here and see what happens.   From browsing your images, it appears to be a challenging dataset.

dusty-nv, 2020-12-01:
> The avg loss appears high (but not INF)

The loss blew up on the second epoch (NaN).   However I am re-running it now with `--learning-rate=0.005` (instead of the default of `0.01`) and so far it is more stable (and the losses are all in normal range now).   It seems due to the difficulty of the dataset, it requires training with a reduced learning rate.

pradhunmya01, 2020-12-02:
> Hi @pradhunmya01 , sorry for the delay while I was catching up from Thanksgiving holiday.
> 
> When I tried to run your dataset, I got a bunch of warning like this:
> 
> ```shell
> warning - image 545-wba15196000c004-1600143490978-top-shelf-0 has object with unknown class 'bottlle'
> warning - image 625-wba15196000c003-1600143407923-bottom-shelf-0 has object with unknown class 'bottlle'
> ```
> 
> These were then causing errors while trying to train the model. In your annotations, there were some typos in the class names. I was able to fix these by running the following commands (make a backup copy of your dataset before running these):
> 
> ```shell
> cd /path/to/your/dataset/Annotations
> sed -i 's/bottlle/bottle/g' *.xml
> sed -i 's/bottled/bottle/g' *.xml
> sed -i 's/Bottlee/bottle/g' *.xml
> ```
> 
> The model then is able to train. I ran it with the default settings:
> 
> ```shell
> python3 train_ssd.py --dataset-type=voc --data=$DATA --model-dir=$MODEL | tee $MODEL/train_log_20201201.txt
> ```
> 
> The avg loss appears high (but not INF), but the avg classification loss is ok (which is the important one to watch). I will let it run here and see what happens. From browsing your images, it appears to be a challenging dataset.

Thank you so much @dusty-nv it's really helpful thanks for your time, Training get started, I'll make sure these things in the future.

pradhunmya01, 2020-12-02:
> The loss blew up on the second epoch (NaN). However I am re-running it now with `--learning-rate=0.005` (instead of the default of `0.01`) and so far it is more stable (and the losses are all in [...]

what do you think other then learning rate what will decrease the loss with this dataset?

closed 2023-04-27 by dusty-nv