dusty-nv/jetson-inference #819: unspecified launch failure (error 4) (hex 0x04)

mayday0208, 2020-11-27:
when i use cudaRGBToRGBAf(cudaRGB.h),cudaRGBAToI420(cudaYUV.h

), return failed.
[image]

my code like this:
[image]

i don't know how to make it right.

mayday0208, 2020-11-27:
__global__ void RGBToRGBAf(uchar3* srcImage,
                           float4* dstImage,
                           uint32_t width,       uint32_t height)
{
    int x, y, pixel;

    x = (blockIdx.x * blockDim.x) + threadIdx.x;
    y = (blockIdx.y * blockDim.y) + threadIdx.y;
	
    pixel = y * width + x;

    if (x >= width)
        return; 

    if (y >= height)
        return;

//	printf("cuda thread %i %i  %i %i pixel %i \n", x, y, width, height, pixel);
		
	const float  s  = 1.0f;
	const uchar3 px = srcImage[pixel];
	
	dstImage[pixel] = make_float4(px.x * s, px.y * s, px.z * s, 255.0f * s);
}

cudaError_t cudaRGBToRGBAf( uchar3* srcDev, float4* destDev, size_t width, size_t height )
{
	if( !srcDev || !destDev )
		return cudaErrorInvalidDevicePointer;

	const dim3 blockDim(8,8,1);
	const dim3 gridDim(iDivUp(width,blockDim.x), iDivUp(height,blockDim.y), 1);

	RGBToRGBAf<<<gridDim, blockDim>>>( srcDev, destDev, width, height );
	
	return CUDA(cudaGetLastError());
}

dusty-nv, 2020-11-27:
Have you tried using the newer cudaConvertColor() function instead?

https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/cuda/cudaColorspace.h#L59

It should skip you some of these steps.

closed 2023-04-27 by dusty-nv