dusty-nv/jetson-inference #820: Unable to retrain an ssd-mobilenet model using voc format input

G4GUO, 2020-11-28:
I have successfully managed to re-train the fruit example in the tutorial but when I try to retrain the picture example with voc files 
the training runs OK but when I then try to convert the model using onnx_export.py I get the following errors.

RuntimeError: Error(s) in loading state_dict for SSD:
size mismatch for classification_headers.0.weight: copying a param with shape torch.Size([30, 512, 3, 3]) from checkpoint, the shape in current model is torch.Size([24, 512, 3, 3]).
size mismatch for classification_headers.0.bias: copying a param with shape torch.Size([30]) from checkpoint, the shape in current model is torch.Size([24]).

This seems to indicate that the mobilenet model I have trained is not the same layer format as the mobilnet model the utility is trying to generate. I have run this example in a docker container and outside a docker container and get the same results. I am not sure what the problem is due to. When I do the training I am using the correct mobilenet base model so I would expect it to generate the new model in the same format, which I would then expect to be in the required format for the onnx utility. 
Any help would be appreciated.

- Charles

P.S I have asked this question in the NVIDIA developers forum as well but I thought this might be a better place to ask it.

dusty-nv, 2020-11-28:
Can you check that you don't have a trailing newline (blank line) at the end of your labels.txt file?    Check both the labels.txt with your dataset and the one with your PyTorch model.

G4GUO, 2020-11-28:
Thanks Dusty, you are a star, that fixed it.

What I have actually done is to modify camera-capture to input images directy from an SDR (Software Defined Radio) which I can then manually label. Eventually I am going to have to figure out how to do the training on my Linux machine which has an RTX3090 and probably train the model from scratch rather than using transfer learning.

Thanks again.

- Charles

dusty-nv, 2020-11-28:
OK great, glad you got it to work.  Can you add a reply to the forum thread linking to this GitHub issue?

BTW you can run the PyTorch SSD code on a Linux PC.  It is a git submodule - https://github.com/dusty-nv/pytorch-ssd

G4GUO, 2020-11-28:
OK will do and thanks again.

closed 2020-11-28 by G4GUO