dusty-nv/jetson-inference #821: Runtime error - train.py

DJT777, 2020-11-29:
Hey dusty,

I am having a runtime error that I am unable to solve on my own. Would you take a look at this output in the terminal when I am attempting to do the training part of the tutorial?

```
dylan@tx1:/media/dylan/SD/jetson-inference/python/training/classification$ sudo python3 train.py --model-dir=models/cat_dog data/cat_dog
Use GPU: 0 for training
=> dataset classes:  2 ['cat', 'dog']
=> using pre-trained model 'resnet18'
=> reshaped ResNet fully-connected layer with: Linear(in_features=512, out_features=2, bias=True)
Traceback (most recent call last):
  File "train.py", line 506, in <module>
    main()
  File "train.py", line 135, in main
    main_worker(args.gpu, ngpus_per_node, args)
  File "train.py", line 277, in main_worker
    train(train_loader, model, criterion, optimizer, epoch, num_classes, args)
  File "train.py", line 334, in train
    acc1, acc5 = accuracy(output, target, topk=(1, min(5, num_classes)))
  File "train.py", line 500, in accuracy
    correct_k = correct[:k].view(-1).float().sum(0, keepdim=True)
RuntimeError: view size is not compatible with input tensor's size and stride (at least one dimension spans across two contiguous subspaces). Use .reshape(...) instead.
```

DJT777, 2020-11-29:
I am getting the same error with the plant training

```
dylan@tx1:/media/dylan/SD/jetson-inference/python/training/classification$ sudo python3 train.py --model-dir=models/plants data/PlantCLEF_Subset
Use GPU: 0 for training
=> dataset classes:  20 ['ash', 'beech', 'cattail', 'cedar', 'clover', 'cyprus', 'daisy', 'dandelion', 'dogwood', 'elm', 'fern', 'fig', 'fir', 'juniper', 'maple', 'poison_ivy', 'sweetgum', 'sycamore', 'trout_lily', 'tulip_tree']
=> using pre-trained model 'resnet18'
=> reshaped ResNet fully-connected layer with: Linear(in_features=512, out_features=20, bias=True)
Traceback (most recent call last):
  File "train.py", line 506, in <module>
    main()
  File "train.py", line 135, in main
    main_worker(args.gpu, ngpus_per_node, args)
  File "train.py", line 277, in main_worker
    train(train_loader, model, criterion, optimizer, epoch, num_classes, args)
  File "train.py", line 334, in train
    acc1, acc5 = accuracy(output, target, topk=(1, min(5, num_classes)))
  File "train.py", line 500, in accuracy
    correct_k = correct[:k].view(-1).float().sum(0, keepdim=True)
RuntimeError: view size is not compatible with input tensor's size and stride (at least one dimension spans across two contiguous subspaces). Use .reshape(...) instead.
****
```

dusty-nv, 2020-11-30:
Hi @DJT777 , which version of PyTorch / torchvision do you have installed?    I haven't seen this error before.

Can you try using the jetson-inference container for training?

DJT777, 2020-11-30:
@dusty-nv 

```
dylan@tx1:~$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> print(torch.__version__)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'torch' is not defined
>>> import torch
>>> print(torch.__version__)
1.7.0
>>> import torchvision
>>> print(torchvision.__version__)
0.2.2
```

I don't know exactly what you mean by using the jetson-inference container for training...?

Something like this...?

```
dylan@tx1:/media/dylan/SD/jetson-inference$ sudo python3 ./python/training/classification/train.py --model-dir=./python/training/classification/models/cat_dog ./python/training/classification/data/cat_dog
Use GPU: 0 for training
=> dataset classes:  2 ['cat', 'dog']
=> using pre-trained model 'resnet18'
=> reshaped ResNet fully-connected layer with: Linear(in_features=512, out_features=2, bias=True)
Traceback (most recent call last):
  File "./python/training/classification/train.py", line 506, in <module>
    main()
  File "./python/training/classification/train.py", line 135, in main
    main_worker(args.gpu, ngpus_per_node, args)
  File "./python/training/classification/train.py", line 277, in main_worker
    train(train_loader, model, criterion, optimizer, epoch, num_classes, args)
  File "./python/training/classification/train.py", line 334, in train
    acc1, acc5 = accuracy(output, target, topk=(1, min(5, num_classes)))
  File "./python/training/classification/train.py", line 500, in accuracy
    correct_k = correct[:k].view(-1).float().sum(0, keepdim=True)
RuntimeError: view size is not compatible with input tensor's size and stride (at least one dimension spans across two contiguous subspaces). Use .reshape(...) instead.
```

dusty-nv, 2020-11-30:
> >>> print(torchvision.__version__)
> 0.2.2

I think that version of torchvision seems pretty old to be using with PyTorch 1.7.   torchvision 0.2.2 was released with PyTorch 1.0.  Whereas PyTorch 1.7 was released with torchvision 0.8.1.   

Note that I haven't tested these training scripts against PyTorch 1.7, so I'm not sure if the error is related to that or your older torchvision.   So you might want to downgrade to PyTorch 1.6 / torchvision 0.7.0

The container already comes with known-working versions of PyTorch and torchvision installed.  You can see the instructions for running the container here:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md

DJT777, 2020-11-30:
@dusty-nv 

I was able to find the Jetson wheel file for PyTorch and Vision
```
>>> print(torchvision.__version__)
0.9.0a0+6e7ed49
>>> print(torch.__version__)
1.7.0
```
Both of these seem to  be working.

If you recall from the first isssue I had, I was having difficulties with the docker container and have built from source. (was posting about the issue with the SD card mounting and formatting like ~2 weeks ago)

I am getting the same output now with updated torchvision and PyTorch...
```

>>> print(torchvision.__version__)
0.9.0a0+6e7ed49
>>> print(torch.__version__)
1.7.0
```

Output from two attempts to run train.py

```
dylan@tx1:/media/dylan/SD/jetson-inference/python/training/classification$ sudo  python3 train.py --model-dir=models/cat_dog data/cat_dog
Use GPU: 0 for training
=> dataset classes:  2 ['cat', 'dog']
=> using pre-trained model 'resnet18'
=> reshaped ResNet fully-connected layer with: Linear(in_features=512, out_features=2, bias=True)
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/queues.py", line 240, in _feed
    send_bytes(obj)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 200, in send_bytes
    self._send_bytes(m[offset:offset + size])
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
Traceback (most recent call last):
  File "train.py", line 506, in <module>
    main()
  File "train.py", line 135, in main
    main_worker(args.gpu, ngpus_per_node, args)
  File "train.py", line 277, in main_worker
    train(train_loader, model, criterion, optimizer, epoch, num_classes, args)
  File "train.py", line 330, in train
    output = model(images)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.9.0a0+6e7ed49-py3.6-linux-aarch64.egg/torchvision/models/resnet.py", line 249, in forward
    return self._forward_impl(x)
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.9.0a0+6e7ed49-py3.6-linux-aarch64.egg/torchvision/models/resnet.py", line 239, in _forward_impl
    x = self.layer3(x)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/container.py", line 117, in forward
    input = module(input)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.9.0a0+6e7ed49-py3.6-linux-aarch64.egg/torchvision/models/resnet.py", line 74, in forward
    out = self.conv2(out)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/conv.py", line 423, in forward
    return self._conv_forward(input, self.weight)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/conv.py", line 420, in _conv_forward
    self.padding, self.dilation, self.groups)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/signal_handling.py", line 66, in handler
    _error_if_any_worker_fails()
RuntimeError: DataLoader worker (pid 18092) is killed by signal: Killed. 
dylan@tx1:/media/dylan/SD/jetson-inference/python/training/classification$ sudo  python3 train.py --model-dir=models/cat_dog data/cat_dog
Use GPU: 0 for training
=> dataset classes:  2 ['cat', 'dog']
=> using pre-trained model 'resnet18'
=> reshaped ResNet fully-connected layer with: Linear(in_features=512, out_features=2, bias=True)
Traceback (most recent call last):
  File "train.py", line 506, in <module>
    main()
  File "train.py", line 135, in main
    main_worker(args.gpu, ngpus_per_node, args)
  File "train.py", line 277, in main_worker
    train(train_loader, model, criterion, optimizer, epoch, num_classes, args)
  File "train.py", line 334, in train
    acc1, acc5 = accuracy(output, target, topk=(1, min(5, num_classes)))
  File "train.py", line 500, in accuracy
    correct_k = correct[:k].view(-1).float().sum(0, keepdim=True)
RuntimeError: view size is not compatible with input tensor's size and stride (at least one dimension spans across two contiguous subspaces). Use .reshape(...) instead.
```

dusty-nv, 2020-11-30:
You might want to downgrade to PyTorch 1.6 and torchvision 0.7.0

DJT777, 2020-12-01:
The tx1 space limitations have been giving me some problems with reinstalling torch and vision. 

After uninstalling them I still have been unable to reinstall from the wheel files and instructions on the Nvidia forums.

I think I will reflash and try to completely rebuild around the space issues.

dusty-nv, 2020-12-01:
If you are going to re-flash, I would just try using the container straight away (hopefully there is enough space at that point)

Otherwise, when you build jetson-inference from source, it will prompt you to install PyTorch and can handle it for you.

dusty-nv, 2020-12-10:
OK, found this issue was because of some updates in PyTorch 1.7 - I fixed it in the classification training script in commit [`15c7d5`](https://github.com/dusty-nv/pytorch-classification/commit/15c7d51cd6a1271e5ee6959c3f91fb95b6b54ff0)

If you are using PyTorch 1.7, re-clone the jetson-inference repo and the classification training should work now.

The updated code should still work on previous versions of PyTorch.

closed 2023-04-27 by dusty-nv