dusty-nv/jetson-inference #822: rtsp have very long delay

AadeIT, 2020-11-30:
Hi， i using the follow command

`python3 detectnet.py rtsp://admin:nncsys2019@192.168.100.91/Streaming/Channels/101?transportmode=unicast
`

There‘s  a long delay compared to rtsp

dusty-nv, 2020-11-30:
Hi @AadeIT , you may try uncommenting this line from [`jetson-inference/utils/codec/gstDecoder.cpp#L521](https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/codec/gstDecoder.cpp#L521)

https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/codec/gstDecoder.cpp#L521

``` cpp
		//ss << " latency=200 drop-on-latency=true";
```

Then re-compile and re-install:

``` bash
$ cd jetson-inference/build
$ make
$ sudo make install
```

AadeIT, 2020-12-01:
Hi @dusty-nv , i am uncommenting  this cpp ,  delay  got better ,thanks for your help.
but i am running 
`python3 segnet.py rtsp://xxx:xxx@192.168.1xx.xx/Streaming/Channels/101?transportmode=unicast --network=fcn-resnet18-cityscapes-1024x512 --input-codec=h264`
result is very bad , output is blur

closed 2023-04-27 by dusty-nv