dusty-nv/jetson-inference #824: I would like to know how to run tlt converter generated .trt engine file in jetson-inference

AhmadShaik, 2020-12-01:
I have trained a detectnet model using transfer learning toolkit. The material in this repo mentioned about inference using onnx and caffe models, I would like to know steps related to .trt engine.

dusty-nv, 2020-12-01:
Hi Shaik, there is an overload of tensorNet::Create() that can load a TRT engine directly.  However, supporting the model would require additional pre/post-processing code to be added like there is in detectNet.cpp for the other models.

AhmadShaik, 2020-12-01:
@dusty-nv Thank you very much for your response. Can you guide me to implement that in python. We can add this to the repo.

dusty-nv, 2020-12-01:
The core pre/post-processing is implemented in C++/CUDA.   For example here is where the pre-processing code starts:

https://github.com/dusty-nv/jetson-inference/blob/8498115a4ed8967c7e23f5ae27b5aa2fa1d1431b/c/detectNet.cpp#L716

The Python bindings simply call into this C++ code.

TheLorma200, 2020-12-10:
> I have trained a detectnet model using transfer learning toolkit. The material in this repo mentioned about inference using onnx and caffe models, I would like to know steps related to .trt engine.

Hi @AhmadShaik, my team and I are facing the same problem.
 If you are working on it i would be interested to know how are you trying to solve it and how it is going. If you need any help we could work on it

dwalkes, 2021-01-10:
Thanks @dusty-nv for the work you've done on this repository, really well done.

I'm also interested in this subject, and noticed the [post on the NVIDIA Forum](https://forums.developer.nvidia.com/t/running-jetsons-detectnet-cpp-with-peoplenet-pruned-model/145313/3?u=danwalkes1) and the TLT [Model Export and Deployment](https://docs.nvidia.com/metropolis/TLT/tlt-getting-started-guide/text/faqs.html#model-export-and-deployment) section which references the need to pre process the image and post process the Tensor after inference.

If I run with PeopleNet and a command line like this:
```
detectnet --model=/jetson-inference/data/resnet34_peoplenet_pruned.etlt_b1_gpu0_fp16.engine --input-blob=input_1 --output-cvg=output_cov/Sigmoid --output-bbox=output_bbox/BiasAdd --labels=/jetson-inference/data/labels_peoplenet.txt  file:///jetson-inference/data/sample_1080p_h264.mp4 file:///jetson-inference/data/detectnet_peoplenet_out_sample_1080p_h264.mp4
```
Where `resnet34_peoplenet_pruned.etlt_b1_gpu0_fp16.engine` is the output of the execution on deepstream and `labels_peoplenet.txt` is coming from the Peoplenet zip file, I get an output like [detectnet_peoplenet.log](https://github.com/dusty-nv/jetson-inference/files/5792195/detectnet_peoplenet.log) running against the `sample_1080p_h264.mp4` and an output video [here](https://drive.google.com/file/d/1ukJ-QgQOjjqEc37s5DezpJTobqgcbT60/view?usp=sharing) which doesn't show the correct output but does show that bag detection (but not people detection) is occurring.  This might not be the best video for people detection with the blurred faces/backs to the camera or perhaps something about the fact that preprocessing is missing means detections aren't working properly.

I'm wondering if anyone can provide any pointers about the best way I could learn how to write the code I'd need to add in order to pre/post-process for a TLT model `.engine` file.  It looks like the preprocess step might be as simple as a call to the right cudaTensorXXX function with the right parameters but it's not obvious to me where these parameters would come from.  For postprocessing I need to know how to map the raw outputs the information necessary to fill in the `Detections` object.

I'm guessing this is related to the [Spec File](https://docs.nvidia.com/metropolis/TLT/tlt-getting-started-guide/text/creating_experiment_spec.html?highlight=dims#) but if so the associations aren't obvious to me.

Any pointers or tips are much appreciated.  Also if this is generally not going to be a project that's possible for someone at less than expert level AI skills to complete that feedback would also be really useful.

dusty-nv, 2021-01-10:
Hi Dan, I don't have support in the jetson-inference library for the TLT models - it would require the addition of the pre/post-processing code to support the models.   Instead I recommend running the TLT models through DeepStream, which is documented on each model's page on NGC.

Adding support for these TLT models is something I've been meaning to do but haven't gotten around to yet.  Also since these models are intended for production use-cases the DeepStream runtime may be a better fit.

dwalkes, 2021-01-11:
Thanks @dusty-nv for the response.

I have successfully run the models in DeepStream and I agree this would be the best way to use the models if I were starting with a blank slate.

A bit more background: I'm trying to leverage the TLT models in an existing application which cannot be easily re-architected to support DeepStream.  So I was looking at whether it would be possible to use TLT models with TensorRT instead as an alternative to a more thourough re-architect of an existing proprietary application in the short term.

Regarding:
>  it would require the addition of the pre/post-processing code to support the models

I understand at a high level what needs to be done but I'm unclear about the possible ways to accomplish this.  Specifically I'm wondering which of the three approaches below could work:

1. Write the code myself, using either some spec I haven't found yet or leveraging some existing DeepStream code I haven't found yet.
2. Fund an AI expert to help write this who would know how to do this based on existing documentation or source and a deeper knowledge than I have about how the tensors are formatted.
3. Wait for you or an NVIDIA employee to do this, since there's not enough publicly available information shared about the tensors and it's not possible/practical to determine this without additional documentation.

I know option 3 is possible based on what you have already mentioned here but I'm wondering if options 1 or 2 are worth pursuing.

dusty-nv, 2021-01-11:
I heard from the TLT team that there is a Python-based tool called `tlt-infer` included in the TLT SDK that runs TLT models with TensorRT.  You could try checking that out and see how the pre/post-processing is done there, and/or adopt that code into your application.

dwalkes, 2021-01-11:
Thanks for the suggestion @dusty-nv 
I gave this a shot with `nvcr.io/nvidia/tlt-streamanalytics:v2.0_py3` and I see when I run this container that the `tlt-infer` is just a python wrapper on `iva.common.magnet_infer` which is a compiled python file.  So unless there's a way to get the original python source for `iva.common.magnet_infer` and friends this looks like it might be a dead end.  I can ask about this on the TLT forum if that's a better place to post this question.

dusty-nv, 2021-01-11:
Hmm ok, yea - they should know more about it on the TLT forums.  Thanks.

dwalkes, 2021-01-11:
OK thank you for your help, I've started a new thread at https://forums.developer.nvidia.com/t/source-code-documentation-for-tlt-infer-and-tlt-tensorrt/165684

closed 2023-04-27 by dusty-nv