dusty-nv/jetson-inference #827: detectnet-camera does not work on DLA with ssd-mobilenet-v1 or v2.

dataplayer12, 2020-12-03:
This is an extension of the discussion from #396. Initially the problem was encountered that when inference was performed on `ssd-mobilenet-v2` using `DEVICE_DLA`, the network didn't detect any objects in the image. I was passing data from a `cv::Mat` as explained by @dusty-nv in #396. He asked me to check if the problem is specific to data passed from OpenCV or not. My tests on Xavier NX show that the problem with DLA persists even when OpenCV is not used. I modified `Create` functions in `detectNet.h` to use the default device as `DEVICE_DLA` and ran `make`. The compiled `detectnet-camera` example does not detect objects with default network `ssd-mobilenet-v2`.  Also, Xavier NX fails to create engine for `ssd-mobilenet-v1` on DLA. So, I checked all the pre-trained models and here are the results for all of them on NX@fp16:

- ssd-mobilenet-v1 : fails to compile engine, error log below.
- ssd-mobilenet-v2 (default): does not detect objects
- ssd-inception-v2: works flawlessly
- pednet: works but with fps of ~15 on Xavier NX, well below expected peformance, TRT logs show all layers running on DLA, none on GPU.
- multiped: same low performance as pednet
- facenet: works flawlessly (~40 fps)
- coco-airplane: works, but at low throughput (19 fps)
- coco-bottle: same as coco-airplane
- coco-chair: same as coco-airplane
- coco-dog: same as coco-airplane

Partial error log for compilation of engine for `ssd-mobilenet-v1`:
```bash
[TRT]    *************** Autotuning format combination: Float(1,1,91,98553), Float(1,1,91,54600), Float(1,1,91,13650), Float(1,1,91,4914), Float(1,1,91,2184), Float(1,1,91,546) -> Float(1,1,1,174447) ***************
[TRT]    *************** Autotuning format combination: Float(1,1,1,7668), Float(1,1,7668,15336), Float(1,1,1,174447) -> Float(1,7,700,700), Float(1,1,1,1) ***************
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```
When monitoring the RAM usage, I noticed that the compiler begins to eat up a lot of memory and eventually runs out, which gives the bad_alloc error. This error occurs only on mobilenet-v1

dusty-nv, 2020-12-03:
Hi @dataplayer12 , DLA doesn't support all models.  You can find the layers that it supports here:

https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html#dla_layers

The pednet/multiped/facenet/coco-* models use an older detection DNN architecture (called 'DetectNet') and their performance is lower even on GPU.

I believe that DeepStream has more comprehensive support for DLA, so you may want to check that out.

dataplayer12, 2020-12-04:
@dusty-nv Thanks. I had hoped that since I enabled gpu fallback, some layers would run on DLA. Why does mobilenet-v2 compile successfully and then not run correctly? Is this something to be on the lookout for when using DLAs with my own models?

closed 2023-04-27 by dusty-nv