dusty-nv/jetson-inference #828: Error when executing "make" for "my-recognition.cpp"

samaujs, 2020-12-03:
In the "Coding Your Own Image Recognition Program (C++)" section, I have encountered the following error : 
$ make
Scanning dependencies of target my-recognition
[100%] Linking CXX executable my-recognition
/usr/lib/gcc/aarch64-linux-gnu/7/../../../aarch64-linux-gnu/Scrt1.o: In function `_start':
(.text+0x18): undefined reference to `main'
(.text+0x1c): undefined reference to `main'
collect2: error: ld returned 1 exit status
CMakeFiles/my-recognition.dir/build.make:75: recipe for target 'my-recognition' failed
make[2]: *** [my-recognition] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/my-recognition.dir/all' failed
make[1]: *** [CMakeFiles/my-recognition.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
--

Some details : 
- Installing on Xavier AGX
- No error for cmake and the previous python section of "my-recognition.py".
- I have previously run the same example on a Jetson Nano and it works without errors.

Any recommendations would be helpful.  Thank you.

dusty-nv, 2020-12-03:
Are you able to build the copy of my-recognition that is in the repo?

https://github.com/dusty-nv/jetson-inference/tree/master/examples/my-recognition

You should build it like so:

``` bash
cd /path/to/your/jetson-inference/examples/my-recognition
mkdir build
cd build
cmake ../
make
```

samaujs, 2020-12-04:
Hi,
  yes, the same error with the repository file together with the recommended steps above (as well as the previously, following the steps in Readme).  I remembered the same steps work when I did it on Jetson Nano sometimes ago.  This error appears when I am now trying with Xaiver AGX.

samaujs, 2020-12-04:
Hi,
  I have resolved the error due to the change of filename instead of "my-recognition" that needs to be update in the "CMakeLists.txt" file.  Thanks.

closed 2020-12-04 by samaujs