dusty-nv/jetson-inference #830: the error of 'jetson-inference/build/aarch64/bin/'

hejun888, 2020-12-03:
hi dusty-nv:
why my fold('jetson-inference/build/aarch64/bin/') is below 
[image]
and not networks and images

dusty-nv, 2020-12-03:
Hi @hejun888 , are you missing the `networks/` and `images/` symbolic link?

These should be setup during the `cmake ../` step when you built the project.    Are you sure your disk is formatted as EXT3 or EXT4?    If you are storing this on a FAT or NTFS drive, it won't be able to make symbolic links.

If it is indeed an EXT3/EXT4 drive, you can make the symlinks yourself like so:

``` bash
cd /path/to/your/jetson-inference/build/aarch64/bin
ln -s /path/to/your/jetson-inference/data/networks networks
ln -s /path/to/your/jetson-inference/data/images images
```

hejun888, 2020-12-04:
dusty-nv，This question is difficult for me，the disk is formatted as EXT4 as shown in the image below
[image]
after executing the cmake ../ many times, there are no images/ and networks/ but as below
[image]

closed 2020-12-04 by hejun888