dusty-nv/jetson-inference #831: Unable to train object detection model

2g1bz0, 2020-12-03:
hi,
I need help.
I'm following the video.
After downloading the image from the open image, I tried to train train_ssd.py and when I ran the command I got an error like this: I think 8GB of memory is enough. Is there anything else wrong?

root@abc-desktop:/jetson-inference/python/training/detection/ssd# python3 train_ssd.py --data=data/fruit --model-dir=models/fruit --batch-size=4 --epochs=1
2020-12-03 08:48:52 - Using CUDA...
2020-12-03 08:48:52 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/fruit', dataset_type='open_images', datasets=['data/fruit'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=1, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-12-03 08:48:52 - Prepare training datasets.
2020-12-03 08:48:52 - loading annotations from: data/fruit/sub-train-annotations-bbox.csv
2020-12-03 08:48:52 - annotations loaded from:  data/fruit/sub-train-annotations-bbox.csv
num images:  2022
2020-12-03 08:48:58 - Dataset Summary:Number of Images: 2022
Minimum Number of Images for a Class: -1
Label Distribution:
	Apple: 1466
	Banana: 604
	Grape: 904
	Orange: 2734
	Pear: 356
	Pineapple: 201
	Strawberry: 3402
	Watermelon: 297
2020-12-03 08:48:58 - Stored labels into file models/fruit/labels.txt.
2020-12-03 08:48:58 - Train dataset size: 2022
2020-12-03 08:48:58 - Prepare Validation datasets.
2020-12-03 08:48:58 - loading annotations from: data/fruit/sub-test-annotations-bbox.csv
2020-12-03 08:48:58 - annotations loaded from:  data/fruit/sub-test-annotations-bbox.csv
num images:  365
2020-12-03 08:48:59 - Dataset Summary:Number of Images: 365
Minimum Number of Images for a Class: -1
Label Distribution:
	Apple: 133
	Banana: 50
	Grape: 151
	Orange: 382
	Pear: 36
	Pineapple: 42
	Strawberry: 343
	Watermelon: 51
2020-12-03 08:48:59 - Validation dataset size: 365
2020-12-03 08:48:59 - Build network.
2020-12-03 08:49:00 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
Traceback (most recent call last):
  File "train_ssd.py", line 309, in <module>
    net.init_from_pretrained_ssd(args.pretrained_ssd)
  File "/jetson-inference/python/training/detection/ssd/vision/ssd/ssd.py", line 119, in init_from_pretrained_ssd
    state_dict = torch.load(model, map_location=lambda storage, loc: storage)
  File "/usr/local/lib/python3.6/dist-packages/torch/serialization.py", line 585, in load
    return _legacy_load(opened_file, map_location, pickle_module, **pickle_load_args)
  File "/usr/local/lib/python3.6/dist-packages/torch/serialization.py", line 772, in _legacy_load
    deserialized_objects[key]._set_from_file(f, offset, f_should_read_directly)
RuntimeError: unexpected EOF, expected 257966 more bytes. The file might be corrupted.

dusty-nv, 2020-12-07:
It seems like your `models/mobilenet-v1-ssd-mp-0_675.pth` file may be corrupt.

Can you try running these commands?

``` bash
$ cd jetson-inference/python/training/detection/ssd
$ wget https://nvidia.box.com/shared/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth -O models/mobilenet-v1-ssd-mp-0_675.pth
```

2g1bz0, 2020-12-08:
Thank you
I didn't get any errors during training on train_ssd.py.
but, after exporting, detect failed. In other words, I tried to detect only one image, but the box did not appear.
Why did detect fail?

//This is a state of learning.

root@abc-desktop:/jetson-inference/python/training/detection/ssd# python3 train_ssd.py --data=data/fruit/ --model-dir=models/fruit/ --batch-size=2 --workers=1 --epoch=1
2020-12-08 14:35:00 - Using CUDA...
2020-12-08 14:35:00 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=2, checkpoint_folder='models/fruit/', dataset_type='open_images', datasets=['data/fruit/'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=1, num_workers=1, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-12-08 14:35:00 - Prepare training datasets.
2020-12-08 14:35:00 - loading annotations from: data/fruit/sub-train-annotations-bbox.csv
2020-12-08 14:35:00 - annotations loaded from:  data/fruit/sub-train-annotations-bbox.csv
num images:  2022
2020-12-08 14:35:06 - Dataset Summary:Number of Images: 2022
Minimum Number of Images for a Class: -1
Label Distribution:
	Apple: 1552
	Banana: 610
	Grape: 1130
	Orange: 2452
	Pear: 396
	Pineapple: 219
	Strawberry: 3275
	Watermelon: 265
2020-12-08 14:35:06 - Stored labels into file models/fruit/labels.txt.
2020-12-08 14:35:06 - Train dataset size: 2022
2020-12-08 14:35:06 - Prepare Validation datasets.
2020-12-08 14:35:06 - loading annotations from: data/fruit/sub-test-annotations-bbox.csv
2020-12-08 14:35:06 - annotations loaded from:  data/fruit/sub-test-annotations-bbox.csv
num images:  365
2020-12-08 14:35:08 - Dataset Summary:Number of Images: 365
Minimum Number of Images for a Class: -1
Label Distribution:
	Apple: 135
	Banana: 43
	Grape: 163
	Orange: 260
	Pear: 44
	Pineapple: 39
	Strawberry: 303
	Watermelon: 61
2020-12-08 14:35:08 - Validation dataset size: 365
2020-12-08 14:35:08 - Build network.
2020-12-08 14:35:08 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2020-12-08 14:35:10 - Took 1.77 seconds to load the model.
2020-12-08 14:35:39 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2020-12-08 14:35:39 - Uses CosineAnnealingLR scheduler.
2020-12-08 14:35:39 - Start training from epoch 0.
/usr/local/lib/python3.6/dist-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/usr/local/lib/python3.6/dist-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
2020-12-08 14:36:25 - Epoch: 0, Step: 10/1011, Avg Loss: 14.5199, Avg Regression Loss 4.4924, Avg Classification Loss: 10.0275
2020-12-08 14:36:30 - Epoch: 0, Step: 20/1011, Avg Loss: 10.1059, Avg Regression Loss 3.9718, Avg Classification Loss: 6.1341
2020-12-08 14:36:37 - Epoch: 0, Step: 30/1011, Avg Loss: 8.9187, Avg Regression Loss 3.1364, Avg Classification Loss: 5.7823
2020-12-08 14:36:47 - Epoch: 0, Step: 40/1011, Avg Loss: 9.7260, Avg Regression Loss 4.2638, Avg Classification Loss: 5.4623
2020-12-08 14:36:53 - Epoch: 0, Step: 50/1011, Avg Loss: 10.9533, Avg Regression Loss 4.6289, Avg Classification Loss: 6.3244
2020-12-08 14:37:00 - Epoch: 0, Step: 60/1011, Avg Loss: 9.1247, Avg Regression Loss 3.5626, Avg Classification Loss: 5.5621
2020-12-08 14:37:05 - Epoch: 0, Step: 70/1011, Avg Loss: 8.5219, Avg Regression Loss 3.3023, Avg Classification Loss: 5.2197
2020-12-08 14:37:09 - Epoch: 0, Step: 80/1011, Avg Loss: 8.5600, Avg Regression Loss 2.8761, Avg Classification Loss: 5.6838
2020-12-08 14:37:14 - Epoch: 0, Step: 90/1011, Avg Loss: 7.5302, Avg Regression Loss 2.6558, Avg Classification Loss: 4.8744
2020-12-08 14:37:18 - Epoch: 0, Step: 100/1011, Avg Loss: 8.6010, Avg Regression Loss 3.5954, Avg Classification Loss: 5.0056
2020-12-08 14:37:28 - Epoch: 0, Step: 110/1011, Avg Loss: 8.1808, Avg Regression Loss 3.0599, Avg Classification Loss: 5.1209
2020-12-08 14:37:33 - Epoch: 0, Step: 120/1011, Avg Loss: 8.0501, Avg Regression Loss 3.0772, Avg Classification Loss: 4.9729
2020-12-08 14:37:37 - Epoch: 0, Step: 130/1011, Avg Loss: 8.1948, Avg Regression Loss 3.1437, Avg Classification Loss: 5.0511
2020-12-08 14:37:47 - Epoch: 0, Step: 140/1011, Avg Loss: 7.4560, Avg Regression Loss 2.6546, Avg Classification Loss: 4.8014
2020-12-08 14:37:52 - Epoch: 0, Step: 150/1011, Avg Loss: 7.7333, Avg Regression Loss 2.9388, Avg Classification Loss: 4.7945
2020-12-08 14:37:57 - Epoch: 0, Step: 160/1011, Avg Loss: 8.2852, Avg Regression Loss 3.2806, Avg Classification Loss: 5.0046
2020-12-08 14:38:01 - Epoch: 0, Step: 170/1011, Avg Loss: 7.5026, Avg Regression Loss 2.6710, Avg Classification Loss: 4.8316
[... 243 lines cut ...]
[TRT]    detectNet -- maximum bounding boxes:  3000
[... cut ...]
[TRT]    Post-Process  CPU   5.87615ms  CUDA   6.10953ms
[TRT]    Total         CPU 142.39532ms  CUDA 142.17302ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[image] imageLoader -- End of Stream (EOS) has been reached, stream has been closed
detectnet:  shutting down...
detectnet:  shutdown complete.
root@abc-desktop:/jetson-inference/python/training/detection/ssd#

dusty-nv, 2020-12-08:
It looks like you only trained it for 1 epoch - you would probably need to train it for more epochs (30 is a good start)

BTW here is my fruit model that I trained for 100 epochs:   https://nvidia.box.com/shared/static/gq0zlf0g2r258g3ldabl9o7vch18cxmi.gz

You can try that first and see if it works on your Jetson.

2g1bz0, 2020-12-09:
Thank you. It worked very well. The box output nicely.

closed 2020-12-09 by 2g1bz0