dusty-nv/jetson-inference #832: error: MESA-LOADER: failed to open swrast

hejun888, 2020-12-04:
Hi,
Thank you for making this great tutorial!
I have installed the project but when I try to run the imagenet-camera.py demo, I got this error
 
[image]

It seems that my OpenGL lin is broken? Have you ever encountered this issue?

Thank you!

dusty-nv, 2020-12-04:
Hmm...is your display connected to your Jetson, or are you running headlessly over SSH?

Are you able to run glxgears or glxinfo?   (you may have to install the mesa-utils package first)

This may repair your OpenGL installation if you install it:  https://github.com/NVIDIA/libglvnd

Alternatively, you could try the jetson-inference container, or if all else fails re-flash your SD card to get OpenGL back.

hejun888, 2020-12-05:
thanks,now, it's OK

closed 2020-12-05 by hejun888

Kshitij09-ag, 2021-03-10:
I'm also facing the same issue and I'm running headlessly but have opened the screen via VNC server

Kshitij09-ag, 2021-03-10:
Thanks for the quick reply @dusty-nv. I tried using rtp but for some reason I was not able to stream that rtp on VLC

dusty-nv, 2021-03-10:
> Thanks for the quick reply @dusty-nv. I tried using rtp but for some reason I was not able to stream that rtp on VLC

VLC can be finiky with RTP.  Try installing GStreamer on your PC instead:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#viewing-rtp-remotely