dusty-nv/jetson-inference #834: Image manipulation with Cuda - questions

marvision-ai, 2020-12-05:
Hey @dusty-nv 

Thank you for adding in the CUDA sections! I love to learn about new concepts. 

While reading about it I became more curious about the positives and negatives to using CUDA for image manipulation. 

Clearly there is a reason to have this within the repo and I would love to learn more about it. 

1. Are there speed improvements over numpy? If so are there any numbers? I am novice to CUDA and want to learn more. 
2. Does preloading the image in the GPU help inference time or is it just for image manipulation? 
3. Can you explain a little more about the benefits of zero copy? You cover it slightly but I want to learn more about this concept. 
4. When would you choose to use this CUDA pipeline over the standard opencv and numpy pipeline? 
5. Can you provide me a quick sample code that shows loading the images coming from opencv into the visa mem? You say to convert the colors pace of the image, but I'm not completely sure about all the steps. 

Thanks a bunch!!

dusty-nv, 2020-12-05:
Hi Marvison,

I don't have CPU vs GPU benchmarks for those routines, but CUDA will be faster because it is using the GPU and parallel processing, whereas numpy is using the CPU.  There is a GPU version of numpy called cupy though.

Having the image in memory accessible from GPU is required for inferencing with TensorRT.  That only operates on CUDA pointers.

Since Jetson has an integrated GPU (i.e. not connected over PCIe), it shares the same memory between CPU and GPU.  So using zeroCopy avoids unnecessary copies between the devices.  The zeroCopy memory is just allocated in a certain way through the CUDA API so that the memory is mapped into both GPU and CPU address spaces.

On Jetson you would use CUDA pipeline when you can (i.e. you have all the functions implemented in CUDA or through some library).  It will give you higher performance than CPU-based numpy/OpenCV way.

For examples of converting between OpenCV and CUDA, see these examples:

https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/cuda-from-cv.py

https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/cuda-to-cv.py

marvision-ai, 2020-12-05:
Hi @dusty-nv ,

That makes sense. Is there considerable overhead to do the CV -> CUDA -> CV conversion if I have image manipulation functions? 

Also with the saving image function, is it faster than saving with cv2? 

This topic is quite fascinating !

closed 2023-04-27 by dusty-nv