dusty-nv/jetson-inference #835: [TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1

naimulhq, 2020-12-05:
I am using the video tutorial on Youtube to develop my first real time object detection. When I try to run the script, the following error show up `[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
`
After the error, the camera never turns on and the terminal keeps on spitting values like this:
After concat removal: 66 layers
[TRT]    Graph construction and optimization completed in 0.045886 seconds.
[TRT]    Constructing optimization profile number 0 [1/1].
[TRT]    --------------- Timing Runner: <reformat> (Reformat)
[TRT]    Tactic: 1002 time 0.747864
[TRT]    Tactic: 0 time 1.23659
[TRT]    Fastest Tactic: 1002 Time: 0.747864
[TRT]    --------------- Timing Runner: <reformat> (Reformat)
[TRT]    Tactic: 1002 time 20.0891
[TRT]    Tactic: 0 time 1.31471
[TRT]    Fastest Tactic: 0 Time: 1.31471

Has anyone come up with the same issue and knows how to fix it? I'm running the script on the Jetson Nano 4GB Development kit with Jetpack 4.4 installed and using the Raspberry Pi Camera V2.

This is the actual script:
```import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("coco-dog", threshold=0.5)
camera = jetson.utils.videoSource("csi://0")      # '/dev/video0' for V4L2
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file

while display.IsStreaming():
        img = camera.Capture()
        detections = net.Detect(img)
        display.Render(img)
        display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))```

dusty-nv, 2020-12-07:
> When I try to run the script, the following error show up `[TRT] Could not register plugin creator - ::FlattenConcat_TRT version 1`

Hi @naimulhq , you can ignore that warning.

> After the error, the camera never turns on and the terminal keeps on spitting values like this:

It is normal the first time that you load a model, it will take some minutes for TensorRT to optimize the model.  It should only happen the first time that you run each model, because the optimized model gets cached to disk.   If you let it run for a while it should complete the optimizations and then start the camera normally.

naimulhq, 2020-12-07:
Hi @dusty-nv 

I did that and everything ran smoothly. Thank you for the help!

closed 2020-12-07 by naimulhq

neilyoung, 2021-01-03:
Had the same issue, was about to complain but found this. Thanks so much for this repo and your video. It really helps a lot to have a first contact with the entire material. 

@dusty-nv Do you have a link list of other videos you made? You mentioned that you want to provide another video w.r.t. PyTorch, is it available somewhere?

Thanks a lot for all your work. Very much appreciated. I came from a Raspberry PI 4B/Coral TPU solution and wanted to check, if the Jetson Nano with its GPU is a competitor. I need to infer from 3 cameras simultaneoulsy, will check the next days, how the Jetson and your project performs here. 

BTW: Is the camera capturer thread-safe?

dusty-nv, 2021-01-04:
Hi @neilyoung , you can find all the screencast videos for Hello AI World here:

https://github.com/dusty-nv/jetson-inference#video-walkthroughs

hgddingjun, 2022-04-20:
Hi@dusty-nv，when i try to run "./imagenet images/orange_0.jpg images/test/output_0.jpg" , the following error show up:
...
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
...
[TRT]    2: [utils.cpp::checkMemLimit::380] Error Code 2: Internal Error (Assertion upperBound != 0 failed. Unknown embedded device detected. Please update the table with the entry: {{1794, 6, 16}, 12653},)
[TRT]    device GPU, failed to build CUDA engine
[TRT]    device GPU, failed to load networks/bvlc_googlenet.caffemodel
[TRT]    failed to load networks/bvlc_googlenet.caffemodel
[TRT]    imageNet -- failed to initialize.
imagenet:  failed to initialize imageNet
Could you help me to solve the problem?