dusty-nv/jetson-inference #838: Usage with torch2trt

chunter18, 2020-12-08:
I have been using torch2trt to export my models, but I have been seeing that my application is spending a significant portion of time copying data back and forth between the CPU and GPU on my Xavier, so I am interested in using the ZeroCopy memory.

However, in my tests, it seems that the TRTModule wont accept the image capsule object provided by jetson-utils. Is this a suported usecase? If not, how can I use my own custom models with the imageNet/tensorNet classes provided by jetson-inference?

Thanks!

dusty-nv, 2020-12-08:
Hi @chunter18 , I believe torch2trt expects to recieve PyTorch tensors, to maintain compatibility with PyTorch.   However you could call `.cuda()` on those tensors first.

Is it only doing the memory transfers before the model and then after the model?   You may be able to manage that better yourself by putting the tensors on GPU yourself before calling the model.

To use it through jetson-inference, you would need to export the model to ONNX.   You can see an example of how this is done for classification models:

https://github.com/dusty-nv/pytorch-classification/blob/5107de352cade326e7aacd44bf8d625b7055fd4e/onnx_export.py

What model are you using?   To support it in jetson-inference, you would want to make sure the pre-processing coefficients match what PyTorch is using (like mean pixel subtraction, normalization, and std deviation):

https://github.com/dusty-nv/jetson-inference/blob/8498115a4ed8967c7e23f5ae27b5aa2fa1d1431b/c/imageNet.cpp#L433

chunter18, 2020-12-08:
Hey @dusty-nv:

My use case is a pretty standard classifier application - frames come in via OpenCV, which I turn into torch tensors and send to the GPU using `.cuda()`, and, the result logits are passed back to the CPU using `.cpu()` for further use. Its actually the `.cpu()` calls that are taking most of the time according to cProfile, but I assume using ZeroCopy memory would save me time both ways. 

I'm using a few ResNet based models - I can easily export them to onnx, but is there an example showing how to load them using the `jetson.inference.imageNet` class? I have only been able to find examples showing the usage of the models downloaded as part of the installation process.

Thanks for the help so far!

dusty-nv, 2020-12-08:
If you check here, it shows the syntax for loading your own classification model from ONNX:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md#processing-images-with-tensorrt
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md#training-your-model

Those tutorials go through training your own model and exporting it to ONNX.  That command line to load your own ONNX model expects the input/output node names to be set the way they are in my `onnx_export.py` script.

chunter18, 2020-12-09:
Thanks Dustin, that's exactly what I needed. Ill close out the issue.

closed 2020-12-09 by chunter18