dusty-nv/jetson-inference #839: Exception: jetson.utils -- failed to create videoSource device

wang911205, 2020-12-08:
when I run  ./detectnet-console.py /usr/share/visionworks/sources/data/pedestrians.mp4 pedestrians_ssd.mp4
I got the wrong:
Traceback (most recent call last):
  File "./detectnet-console.py", line 54, in <module>
    input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device

dusty-nv, 2020-12-08:
Hi @wang911205 , is there any additional terminal log above the exception?

Can you post the full console output?  Thanks.

wang911205, 2020-12-09:
no others!

dusty-nv, 2020-12-09:
Can you try running it like this, and then post the full terminal log?

``` bash
python3 detectnet.py /usr/share/visionworks/sources/data/pedestrians.mp4 pedestrians_ssd.mp4
```

wang911205, 2020-12-10:
> Can you try running it like this, and then post the full terminal log?
> 
> ```shell
> python3 detectnet.py /usr/share/visionworks/sources/data/pedestrians.mp4 pedestrians_ssd.mp4
> ```

This code can run well.

wang911205, 2020-12-10:
I have a problem with the device connected with the remote tool.

dusty-nv, 2020-12-10:
> I have a problem with the device connected with the remote tool.

Do you mean when you are running your Jetson headlessly, without a display attached?     It won't show the OpenGL window then, it can't be tunneled through SSH.

mjcov, 2021-01-18:
Hello,

I'm having a similar problem where using an RTSP stream as my source in the terminal for both video-viewer and detectnet, everything works fine.  However when I create a Python script using the same stream info it errors:

`URI -- missing/invalid IP port from rtsp://admin:XXXX@192.168.20.121, default to port 554
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:XXXX@192.168.20.121
Traceback (most recent call last):
  File "/home/mjcov/Desktop/pyPro/Jetson Inference/DetectHumans1.py", line 6, in <module>
    camera = jetson.utils.videoSource("rtsp://admin:XXXX@192.168.20.121")
Exception: jetson.utils -- failed to create videoSource device
mjcov@mjcov-desktop:~/Desktop/pyPro

Thank you in advance for any help.

dusty-nv, 2021-01-18:
Hmm. If you replace opt.output_URI in detectnet.py with your RTSP string, does it still work?

https://github.com/dusty-nv/jetson-inference/blob/5718df1495859808d5d076efea597b7e641bcb96/python/examples/detectnet.py#L55

mjcov, 2021-01-18:
Hey Dusty,

Thanks for the quick response.  I tried your suggestion but got nowhere.  I added a codec arg based on a suggestion you made to someone else and I got it to recognize the camera.  However, that led to another issue below:

[OpenGL] failed to create X11 Window.
jetson.utils -- no output streams, creating fake null output

I ended up finding this:  https://github.com/dusty-nv/jetson-inference/issues/619

This fix worked, but I'm curious if this is the final workaround, or if there is something a bit less sketchy that will correct this?

Thanks again for any input.

dusty-nv, 2021-01-18:
Rather than changing the imports, you should just be able to create the videoOutput before the videoSource (and not need to change the order of the imports).   Unfortunately I haven't been able to reproduce and debug this issue on my system, so haven't been able to find a better solution - sorry about that.