dusty-nv/jetson-inference #840: git submodule update --init no result

yuyuanc-chen, 2020-12-09:
Hi. I was building the project from source.I totally do the every step.But when I input the command git submodule update --init,there is no response.Then I input cmake ../  and it occures :required git submodules have not been detected.First,please run the git submodule update --init  from the project's roo directory to clone them.But I run the comman for servral times without any response.I don't know how to solve it.

dusty-nv, 2020-12-11:
Hmm ok.   Can you try re-cloning the repo with the `--recursive` flag, like below?

``` bash
$ git clone --recursive https://github.com/dusty-nv/jetson-inference
```

This will automatically grab the submodules for you, so you won't need the `git submodule` step then.

yuyuanc-chen, 2020-12-12:
Thank you.I can't download it by git clone xx because vpn doesn't work. I download it on the github directely.

closed 2023-04-27 by dusty-nv