dusty-nv/jetson-inference #842: Training custom dataset error

ammarjafri, 2020-12-10:
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/queues.py", line 234, in _feed
    obj = _ForkingPickler.dumps(obj)
  File "/usr/lib/python3.6/multiprocessing/reduction.py", line 51, in dumps
    cls(buf, protocol).dump(obj)
_pickle.PicklingError: Can't pickle <class 'MemoryError'>: it's not the same object as builtins.MemoryError

I'm training a custom model and getting this error. I have used open_images_downloader.py to dowload dataset for mobile phone and seat belt, and it have 4700 pics.

dusty-nv, 2020-12-11:
Can you keep an eye on `tegrastats` and `dmesg --w` while running this to see if you are running out of memory?

If so, you can mount additional SWAP and if needed disable the desktop GUI:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap

You can also reduce the `--batch-size` and `--workers` to decrease the memory usage.  Since this error seems related to loading the data, you may want to reduce the workers (dataloader threads) or turn off dataloader threading with `--workers=0`

ammarjafri, 2020-12-15:
@dusty-nv thanks that works. I have trained a model using over 100 epochs, but unfortunately the result is not good, it randomly detects anything to mobile in the screen. Please can suggest me how to improve this and get better result.
[image]
[image]
[image]
[image]
[image]

dusty-nv, 2020-12-15:
Hi @ammarjafri , glad you were able to get it training.   Does it mostly detect monitors/TV's as mobile phones?    I suppose those look fairly close.  Did you dataset contain images of monitors/TV's as well, so that the model can learn those _aren't_ mobile phones?  

If you run the model inferencing on your training images, how does it do there?

ammarjafri, 2020-12-16:
> Hi @ammarjafri , glad you were able to get it training. Does it mostly detect monitors/TV's as mobile phones? I suppose those look fairly close. Did you dataset contain images of monitors/TV's as well, so that the model can learn those _aren't_ mobile phones?
>
@dusty-nv Thanks for you to reply that fast, No it even detect faces to mobile phone. The dataset I use is from Open Images Dataset using open_images_downloader.py and I have check the dataset it does not contain mointors/TV's.

> If you run the model inferencing on your training images, how does it do there?
Yes, I have tried to make a model on training on my own images, but the result was about same, it also randomly detects mobile phone on face, TV or any other object.

Don't know what is problem that causing that randomly detects my object on any object, do you have any idea why I could occur?

dusty-nv, 2020-12-16:
Can you delete the .engine file from your model's directory and try running detectnet program again?

You can also try using the [`run_sdd_example.py`](https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/run_ssd_example.py) script to test the PyTorch version of the model on some test images to see if it does better.

ammarjafri, 2020-12-29:
> Can you delete the .engine file from your model's directory and try running detectnet program again?
>
> You can also try using the [...]

Tried everything, but accuracy is not improving I tried to make with my own dataset, after 2500 pics I get a little detection, but still it was not enough, it could not detect from far distance

closed 2023-04-27 by dusty-nv