dusty-nv/jetson-inference #844: improve scrolling display inference results in cli so it sstatic

Niko-La, 2020-12-11:
Running python detectnet.  If the follow cli output can update without scrolling sporatically it will make it much easier to see the table in action.  

Right now its only readable once exiting the main script
~~~~
detected 0 objects in image                                                     
                                                                                
[TRT]    ------------------------------------------------                       
[TRT]    Timing Report /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_vf
[TRT]    ------------------------------------------------                       
[TRT]    Pre-Process   CPU   0.05271ms  CUDA   1.00693ms                        
[TRT]    Network       CPU  49.81824ms  CUDA  40.28688ms                        
[TRT]    Post-Process  CPU   0.04323ms  CUDA   0.04349ms                        
[TRT]    Visualize     CPU   0.17954ms  CUDA   1.73354ms                        
[TRT]    Total         CPU  50.09371ms  CUDA  43.07084ms                        
[TRT]    ------------------------------------------------                       
                                                                                
detected 0 objects in image                                                     
                                                                                
[TRT]    ------------------------------------------------                       
[TRT]    Timing Report /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_vf
[TRT]    ------------------------------------------------                       
[TRT]    Pre-Process   CPU   0.05281ms  CUDA   0.98672ms                        
[TRT]    Network       CPU  51.04545ms  CUDA  41.72052ms                        
[TRT]    Post-Process  CPU   0.04682ms  CUDA   0.04677ms                        
[TRT]    Visualize     CPU   0.17954ms  CUDA   1.73354ms                        
[TRT]    Total         CPU  51.32463ms  CUDA  44.48755ms                        
[TRT]    ------------------------------------------------

~~~~

dusty-nv, 2020-12-11:
Hi @Niko-La, try commenting out lines 66-69 of detectnet.py:

https://github.com/dusty-nv/jetson-inference/blob/1f3c3ca1e5ae2ffd3c608f3711bf251e89e1ec08/python/examples/detectnet.py#L66

Niko-La, 2020-12-11:
@dusty-nv  just tried it, it still has the table scrolling.  I have seen some script in other applications display it in the format sticky with the changing numbers updating without the scrolling.

closed 2023-04-27 by dusty-nv