dusty-nv/jetson-inference #845: detectnet fails.

2g1bz0, 2020-12-12:
翻訳がおかしかったら、ごめんなさい
dockerを使っていません。
"test"ディレクトリの下に"jetson-inference"ディレクトリを作り、ビルドをしました。

学習(traffic light, traffic sign)と、エキスポートは、成功しました。
detectnetをrunしようとしました。しかし、以下のようなエラーが発生しました。助けてもらえますか？

[detectnetのコマンドをターミナルで打ちました。
tabボタンの補完は、失敗していました。
私はそこが怪しいと思っています。]

abc@abc-desktop:~/test/jetson-inference/python/training/detection/ssd$ detectnet --model=models/traffic_1/ssd-mobilenet.onnx --labels=models/traffic_1/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "/test/jetson-inference/data/images/city_0.jpg" /test/jetson-inference/data/test/city_0/jpg
```
listDir('/test/jetson-inference/data/images/city_0.jpg') - found no matches
[image] imageLoader -- failed to find '/test/jetson-inference/data/images/city_0.jpg'
detectnet:  failed to create input stream
```

2g1bz0, 2020-12-18:
I'm totally bad. sorry. The command was wrong.
There was wasted space and the directory was specified incorrectly.
After fixing it, detectnet worked.
Also, when specifying model options, is tab completion impossible?

[Right]
detectnet --model=models/traffic_1/ssd-mobilenet.onnx --labels=models/traffic_1/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "~/test/jetson-inference/data/images/city_0.jpg" ~/test/jetson-inference/data/images/test/city_0.jpg

closed 2020-12-18 by 2g1bz0

GT1235, 2021-07-28:
Hi @2g1bz0 , how did you solve this problem? I am also stuck on this exact problem.

dusty-nv, 2021-07-28:
Hi @GT1235, are you sure the path to the image is correct?  Can you provide the command line you are running?

GT1235, 2021-07-31:
@dusty-nv I followed the instructions for re-training the SSD-Mobilenet v2 via this website: https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md
The line I ran was: 
detectnet --model=models/vegetable/ssd-mobilenet.onnx --labels=models/vegetable/labels.txt \
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            "$IMAGES/vegetable_*.jpg" $IMAGES/test/vegetable_%i.jpg
(Copied exactly from the website.)
When I go into the vegetables folder, the images are named with random symbols and letters. I assume that at some point in the process, the test images should be labeled as "vegetable_x.jpg" or something along those lines. However, that is not happening for me at all...

The line I ran before was:    IMAGES=/home/gauri-nano/jetson-inference/data/images

dusty-nv, 2021-07-31:
@GT1235 what did you set `IMAGES` environment variable to?

If your images are randomly named, you can pass it `"$IMAGES/*.jpg" instead

GT1235, 2021-07-31:
@dusty-nv Exactly what line should I run instead? Sorry, I am a beginner.

dusty-nv, 2021-07-31:
OK, instead take out references to $IMAGES and replace it with the actual path to your imagea