dusty-nv/jetson-inference #846: "display://0" does not open preview when running detect.py using rtsp input

Blu3DrAEm, 2020-12-15:
Hello!
I am using a camera from an RTSP stream and can get the object detection model to work, but it does not display a preview from the camera on the jetson. Is there anything missing from the code I'm using? I have been browsing the forums but havent been able to come up with a solution just yet. I will post the code I'm using below. 
P.s. I am not running this inside of the docker container and was able to verify object detection was being performed when I set the output to 'file.mp4', which saved a video file once I stopped the script. I appreciate any help! :)

'''
import jetson.inference
import jetson.utils

import argparse
import sys

# parse the command line
parser = argparse.ArgumentParser(description="Locate objects in a live camera stream using an object detection DNN.", 
                                 formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.detectNet.Usage() +
                                 jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("input_URI", type=str, default="", nargs='?', help="URI of the input stream")
parser.add_argument("output_URI", type=str, default="", nargs='?', help="URI of the output stream")
parser.add_argument("--network", type=str, default="ssd-mobilenet-v2", help="pre-trained model to load (see below for options)")
parser.add_argument("--overlay", type=str, default="box,labels,conf", help="detection overlay flags (e.g. --overlay=box,labels,conf)\nvalid combinations are:  'box', 'labels', 'conf', 'none'")
parser.add_argument("--threshold", type=float, default=0.5, help="minimum detection threshold to use") 

is_headless = ["--headless"] if sys.argv[0].find('console.py') != -1 else [""]

try:
	opt = parser.parse_known_args()[0]
except:
	print("")
	parser.print_help()
	sys.exit(0)

# load the object detection network
net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)

# create video sources & outputs
input = jetson.utils.videoSource('rtsp://192.168.0.47', argv=['--input-codec=h264']) 
output = jetson.utils.videoOutput("display://0")  # this saves a video file with successful object detection if I change to 'file.mp4'   

# process frames until the user exits
while True:
	# capture the next image
	img = input.Capture()

	# detect objects in the image (with overlay)
	detections = net.Detect(img, overlay=opt.overlay)

	# print the detections
	print("detected {:d} objects in image".format(len(detections)))

	for detection in detections:
		print(detection)

	# render the image
	output.Render(img)

	# update the title bar
	output.SetStatus("{:s} | Network {:.0f} FPS".format(opt.network, net.GetNetworkFPS()))

	# print out performance info
	net.PrintProfilerTimes()

	# exit on input/output EOS
	if not input.IsStreaming() or not output.IsStreaming():
		break

dusty-nv, 2020-12-16:
When using RTSP, can you swap the order of the output and input creation to the following?

``` python
output = jetson.utils.videoOutput("display://0") # this saves a video file with successful object detection if I change to 'file.mp4'
input = jetson.utils.videoSource('rtsp://192.168.0.47', argv=['--input-codec=h264'])
```

Blu3DrAEm, 2020-12-16:
> When using RTSP, can you swap the order of the output and input creation to the following?
>
> ```python
> output = jetson.utils.videoOutput("display://0") # this saves a video file with successful [...]

Hey Dusty! That solution worked. Thank you so much! I really appreciate your help!

closed 2020-12-16 by Blu3DrAEm