dusty-nv/jetson-inference #849: docker/run.sh error about permission

ingbeeedd, 2020-12-17:
[image]

ingbeeedd, 2020-12-17:
In addition, ssd wget failed in model folder due to permission issue

dusty-nv, 2020-12-17:
What version of JetPack are you running?   It should be JetPack 4.4.1 (L4T R32.4.4) to run the jetson-inference:r32.4.4.   There is also a container tag for JetPack 4.4 (L4T R32.4.3), jetson-inference:r32.4.3.

The folder permissions seem unrelated - you can run wget with sudo or change the folder permissions with `chmod -R 666 <directory>`

ingbeeedd, 2020-12-21:
I've only run a Docker script, and that directory has root privileges. When checking nv_tegra_release, it is 4.4.

ingbeeedd, 2020-12-21:
I just installed the github contents after installing the jetpack, so I don't think there will be any conflicts.

ingbeeedd, 2020-12-21:
It's good to do it on the server, not the docker client.

closed 2023-04-27 by dusty-nv