dusty-nv/jetson-inference #851: How to compute the loss function curve from the train.py in image classification

danisomo, 2020-12-18:
Hello, first of all, i am doing my final project of Master's Degree and i had some issues which i can't solve.
My project is about the jetbot recognising four classes images in a long corridor which are right, left, forward and obstacle and making it move depending on the returned class given by a DNN.
Having said this, i tried using different networks provided such as googlenet, alexnet, resnet18, resnet50, etc. but the results with the pretained models are not very good.
I guess this happens because the classes with which they have trained those pretrained models of these networks have nothing to do with the classes that i am trying to classify, so it does not work, specially with classes right and left, so the next step i followed was to deactivate the flag pretrained in the script of training called train.py writting False in the default value in order to avoid the transfer learning and training the network right from the start. 
Then, i have been analyzing the code and watching the results that appear in terminal but the loss value does not seem very real because it is not appreciated how the losses are decreasing. 
Can anybody help me or explain me why i cannot see the losses converge to a low value as it should happen? The values of the epoch one are very similar in order of magnitude to the last epoch as can be seen in the next pictures:

[image]
[image]

My teacher told me that it is absolutely necessary to represent the loss function curve in the report in order to justify how the network is learning and not only the plot of the accuracy (this one can be plot without difficulties)

dusty-nv, 2020-12-18:
Those loss values are printed for each 10 batches, and vary from batch to batch.   The per-batch accuracy varies up & down also. 
 What you may want to do is print out the loss after the validation step at the end of each epoch.   You could print that here:

https://github.com/dusty-nv/pytorch-classification/blob/15c7d51cd6a1271e5ee6959c3f91fb95b6b54ff0/train.py#L395

The validation loss is tracked in the `losses` object, and you could print them just like Acc@1 and Acc@5

danisomo, 2020-12-21:
Thank you very much @dusty-nv 
For tracking the losses in validation and test is like you said:

[image]
[image]

In order to plot those values i put them in a list as seen in previous photos. Then i call the function plot at the end of the program as seen in next picture:

[image]

The plot function is defined like this:

[image]

At the end i get the results of acc1, acc5 and losses each time an epoch occurs in train and validation.

And when the last epoch is executed i got this two plots which i think they look very good. 

[image]
[image]

Sorry for copying things that should be known but just wanted to know if the values are well obtained for the plots and if the plots look pretty good? 

Thanks you beforehand

closed 2023-04-27 by dusty-nv