dusty-nv/jetson-inference #853: boxes are not detected

2g1bz0, 2020-12-19:
I ran train, export and detect without using docker.
The model trained 2500 images (trafiic_sign, traffic_light). It is 100 epochs.

However, the box is not detected.
I have a question here.
・ Does this model need further learning?
・ What is the best number of batches?
・ I would like to see a graph showing how the correct answer rate is improved by verification. Is this possible?

ssd-mobilenet.onnx
https://drive.google.com/drive/folders/1wOMM7RlTwXGbFANknIcqiYLkKA3UuYsL?usp=sharing

dusty-nv, 2020-12-21:
Can you try deleting the `*.engine` file from your model's directory and re-running the detectnet program?

You can try using PyTorch to process an image with your PyTorch model to verify that it works fine on the PyTorch side.   You can try the `run_ssd_example.py` script:

https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/run_ssd_example.py

2g1bz0, 2020-12-23:
I removed the *engine and detected it again, but only the photo was output. There was no box in the photo. Below is the code at that time.

```
abc@abc-desktop:~/test/jetson-inference/python/training/detection/ssd$ sudo detectnet --model=models/traffic_1/ssd-mobilenet.onnx --labels=models/traffic_1/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "~/test/jetson-inference/data/images/traffic_sign.jpg" ~/test/jetson-inference/data/test_1.jpg
[sudo] password for abc: 
[video]  created imageLoader from file://~/test/jetson-inference/data/images/traffic_sign.jpg
------------------------------------------------
imageLoader video options:
------------------------------------------------
  -- URI: file://~/test/jetson-inference/data/images/traffic_sign.jpg
     - protocol:  file
     - location:  ~/test/jetson-inference/data/images/traffic_sign.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     input
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[video]  created imageWriter from file:///home/abc/test/jetson-inference/data/test_1.jpg
------------------------------------------------
imageWriter video options:
------------------------------------------------
  -- URI: file:///home/abc/test/jetson-inference/data/test_1.jpg
     - protocol:  file
     - location:  /home/abc/test/jetson-inference/data/test_1.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     output
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1920
  -- height:     1080
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------

detectNet -- loading detection network model from:
          -- prototxt     NULL
[... 99 lines cut ...]
[TRT]    Network       CPU 120.78058ms  CUDA 118.44067ms
[TRT]    Post-Process  CPU   1.45568ms  CUDA   1.73281ms
[TRT]    Total         CPU 122.32130ms  CUDA 122.12036ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[image] imageLoader -- End of Stream (EOS) has been reached, stream has been closed
detectnet:  shutting down...
detectnet:  shutdown complete.

```
--------------------------------------------------------------------------------------------------------------------------
I'm a beginner, so about pytorch
So when I run run_ssd_example.py, I get an error. Can you give me a specific command?
```
abc@abc-desktop:~/test/jetson-inference/python/training/detection/ssd$ python3 run_ssd_example.py mb1-ssd-lite --models=models/traffic_1/ssd-mobilenet.onnx --labels=models/traffic_1/labels.txt "/home/abc/test/jetson-inference/data/images/traffic_sign.jpg" ~/test/jetson-inference/data/images/test3.jpg
Traceback (most recent call last):
  File "run_ssd_example.py", line 19, in <module>
    class_names = [name.strip() for name in open(label_path).readlines()]
FileNotFoundError: [Errno 2] No such file or directory: '--labels=models/traffic_1/labels.txt'

```
Thank you in advance.

moritz157, 2021-01-02:
I had the same problem but I noticed, that the default threshold value of detectnet was 0.5. When I set it to a much lower value I started to see boxes in the output. Perhaps my model isn't trained well enough yet.

To lower the output threshold try to use the following flag in your `detectnet`-command: `--threshold=0.05`. You may replace '0.05' with any value between 0 and 1. 

For learning more about the available detectnet-options, I just used `detectnet --help`

2g1bz0, 2021-01-04:
> I had the same problem but I noticed, that the default threshold value of detectnet was 0.5. When I set it to a much lower value I started to see boxes in the output. Perhaps my model isn't trained [...]

Thanks to this, I was able to output the box.
So it was caused by low accuracy.
Indeed, 22% of the traffic lights were detected.
I was convinced that I needed to TRAIN more!

closed 2021-10-15 by 2g1bz0