dusty-nv/jetson-inference #855: Use multiple cameras with jetson inference on nvidia jetson nano

FedericoLanzani, 2020-12-19:
I've just get started using the **jetson nano (4gb)** and I'm doing some "experiments" with this library.

I was wondering if there is a way to **use multiple webcams as input streams for inference**.

- **_If yes_** what is this "magic method"?
- **_If not_** what is the best way to **manage multiple input streams** on jetson nano?

My goal is to perform the object detection provided by the repo on multiple webcams.

Thank you for the help!

WaldoPepper, 2020-12-20:
Dusty has provide a [ROS integration](https://github.com/dusty-nv/ros_deep_learning) of this, so you might take a look.

Using somthing like ROS adds a level of abstraction which one can benefit from a lot. It of course has quite a steep learning curve but the effort is well spent.

I am not exactly sure about a possible multicamera setup, but the code of the ROS nodes mentioned above is clearly written and can easily be understood.

dusty-nv, 2020-12-20:
You could also just create multiple videoSource interfaces in a Python/C++ app an do it that way.  The examples show one camera but you could do multiple.

Then command-line parsing in the examples is setup for one camera, but you can hardcode it or do your own command-line parsing.

FedericoLanzani, 2020-12-20:
> You could also just create multiple videoSource interfaces in a Python/C++ app an do it that way. The examples show one camera but you could do multiple. Then command-line parsing in the examples is setup for one camera, but you can hardcode it or do your own command-line parsing.
> […](#)

Thank you for your time! 
I will try to modify the python code and use multiple IP cameras.

tkislan, 2021-01-02:
I'm currently running this against 3 rtsp camera streams, and it works without a problem.
Every camera feed is read in a loop in a separate thread, and then I push images into a single `queue` for detectnet inference

FedericoLanzani, 2021-01-04:
I've worked a while on this task and I discovered that opencv support gstreamer and this makes the management of multiple cameras easy and fast.

Jetpack comes with a version of opencv 4.4.1 built with gstreamer support.

neilyoung, 2021-01-05:
> I'm currently running this against 3 rtsp camera streams, and it works without a problem.
> Every camera feed is read in a loop in a separate thread, and then I push images into a single `queue` for detectnet inference

I was trying something similar, but with 3 USB cams. My Jetson was powered with a micro USB plug (Raspberry PI). With 3 cams the Jetson crashed out of the sudden all the time. With two cams it worked only if I did the object detection with two different (but from the model identical) networks. 

But I suppose it is the power supply, which is problematic here. I wouldn't really see a need for having separate detection networks for different cams, since there should be no dependency between images.

neilyoung, 2021-01-05:
> I've worked a while on this task and I discovered that opencv support gstreamer and this makes the management of multiple cameras easy and fast.
> 
> Jetpack comes with a version of opencv 4.4.1 built with gstreamer support.

Intersting. You are using OpenCV in conjunction with this package? How do you convert a `Mat` object into the format, this lib requires?

Other than that I was pretty happy with the gstreamer support already provided by this package, so opening a camera as "/dev/videox" is definitely involving gstreamer too, w/o a need to use OpenCV explicitly, IMHO.

FedericoLanzani, 2021-01-05:
> Intersting. You are using OpenCV in conjunction with this package? How do you convert a `Mat` object into the format, this lib requires?
>
> Other than that I was pretty happy with the gstreamer [...]

Hi! Yes I'm using opencv to read and display the images from the cameras that I have connected to the jetson nano. (currently 2/3 webcam but soon I want to use multiple ip cameras).
To perform detection with jetson inference I convert the image that opencv reads (ndarray format if I remember right), to cuda image using the method `jetson.utils.cudaFromNumpy(img)` and then I perform detection usign `model.detect(cuda_img)` with detecnet and a custom model trained on my own.

My pipeline is:
- Getting the image from each camera using opencv built with gstreamer (it comes with jetpack). This task is performed on a separate thread for each camera.
- Convert the images that opencv gives to me using `cuda_img = jetson.utils.cudaFromNumpy(img)`.
- Then perform detection with Detecnet loaded with a custom model.
- I print the output with opencv imshow.

I've just some experience using python and opencv, I'm NOT an expert with gstreamer, jetson inference and nvidia stuff. I'm learning this stuff right now, so I'm sure that there is a better way to write my implementation. 

You are right when you say that this library has already a gstreamer backend implementation, but I had some troubles to make that works efficiently with multiple cameras and multiple output streams in the way that I want. 

If you have some tips to give I'd be very happy!

FedericoLanzani, 2021-01-05:
> I'm currently running this against 3 rtsp camera streams, and it works without a problem.
> Every camera feed is read in a loop in a separate thread, and then I push images into a single `queue` for detectnet inference

Hi! When I use this library for reading images I don't come with the result that I expect, but I didn't try exactly your implementation, maybe I could give it a shot.

How many fps you get with 3 cameras?
Because now with my implementation I get 15 fps with 3 usb webcams + detection + output.
If yours do it better I will do more tests with jetson inference library!

neilyoung, 2021-01-05:
Yes, 15 fps is what I get too with 3 cams (as long as it runs) and with the core implementation of this github. No OpenCV involved. My code (currently just test) is not even multithreaded.  You can find the most recent snippet here https://github.com/dusty-nv/jetson-inference/issues/882#issuecomment-753987388 (2 USB cams only currently, since it crashes most likely due to power issues with 3 cams pretty soon)

With 2 cams the inference engine is giving about 24 fps, so 12 for each. I guess if you would need more fps, you would consider using a Coral USB TPU

FedericoLanzani, 2021-01-06:
> Yes, 15 fps is what I get too with 3 cams (as long as it runs) and with the core implementation of this github. No OpenCV involved. My code (currently just test) is not even multithreaded. You can find the most recent snippet here [#882 (comment)](https://github.com/dusty-nv/jetson-inference/issues/882#issuecomment-753987388) (2 USB cams only currently, since it crashes most likely due to power issues with 3 cams pretty soon)
> 
> With 2 cams the inference engine is giving about 24 fps, so 12 for each. I guess if you would need more fps, you would consider using a Coral USB TPU

Hi! 
Have you ever used coral usb with a rpi? 
Is the coral documentation good?

neilyoung, 2021-01-06:
Well, this is probably not the right place to advertise Coral, but I can encourage you to spend the € 70 in order to purchase one. There is a very nice and well working tutorial and repo https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/blob/master/Raspberry_Pi_Guide.md 

And it does really 20 fps @ 1280x720 with 3 USB cams. 35 fps @ 640x480. You might have a little harder time with RTSP feeds, but since you have already solved this outside this project, maybe not.

In fact, I also just have maybe 10 days of Jetson experience and the main motivation was to check, if the GPU can compete with a Coral TPU. W.r.t. fps it seems it can't. I'm not yet sure w.r.t. the accuracy of detection. Since Coral runs Tensorflow Lite (8 bit floating point accuracy) I would think, it is not that good as with Tensorflow. But I would be happy to share my results with you, once you have tested the Google thing. What discourages me a bit with the Coral is 

a) the extra device
b) the possible landing on Google's graveyard https://killedbygoogle.com/ (if it is not selling as Google expects it).

Other than that it just needs a 2GB RPI 4B, which in the end makes not much difference in price, but I all the time find the RPI less complex compared to the Jetson. And the Coral just needs 2.5 W.

Don't try with RPI 3 or 3A+ or 3B+. Can't compete
Let me know

closed 2023-04-27 by dusty-nv