dusty-nv/jetson-inference #856: Failed to save test images to a folder

SAIL-LAB155, 2020-12-20:
Hi. After I trained a cat-dog classification model and was going to test a folder of images using the following command:

_**python3 imagenet.py --model=../training/classification/models/cat_dog/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=../training/classification/data/cat_dog/labels.txt ../training/classification/data/cat_dog/test/cat ../training/classification/data/cat_dog/test/test_cat_result**_

An error happened:

_[cuda]   registered openGL texture for interop access (500x334, GL_RGB8, 501000 bytes)
[image] invalid extension format './training/classification/data/cat_dog/test/test_cat_result' saving image '../training/classification/data/cat_dog/test/test_cat_result'
[image] valid extensions are:  JPG/JPEG, PNG, TGA, BMP.
[image] imageWriter -- failed to save '../training/classification/data/cat_dog/test/test_cat_result'
Traceback (most recent call last):
  File "/usr/local/bin/imagenet.py", line 77, in <module>
    output.Render(img)
Exception: jetson.utils -- videoOutput failed to render image_

The paths are right because I successfully in inferring images without assigning the output folder.

dusty-nv, 2020-12-21:
Can you run `mkdir ../training/classification/data/cat_dog/test/test_cat_result` first?

SAIL-LAB155, 2020-12-22:
Yes, I have created the folder before evaluating. But this error still occurs.
The following information is shown below.

**hkuit155@hkuit155-desktop:~/jetson-inference/python/examples$ mkdir ../training/classification/data/cat_dog/test/test_cat_result
mkdir: cannot create directory ‘../training/classification/data/cat_dog/test/test_cat_result’: File exists
hkuit155@hkuit155-desktop:~/jetson-inference/python/examples$ python3 imagenet.py --model=../training/classification/models/cat_dog/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=../training/classification/data/cat_dog/labels.txt ../training/classification/data/cat_dog/test/cat ../training/classification/data/cat_dog/test/test_cat_result
jetson.inference -- imageNet loading network using argv command line params

imageNet -- loading classification network model from:
         -- prototxt     (null)
         -- model        ../training/classification/models/cat_dog/resnet18.onnx
         -- class_labels ../training/classification/data/cat_dog/labels.txt
         -- input_blob   'input_0'
         -- output_blob  'output_0'
         -- batch_size   1

[TRT]    TensorRT version 7.1.3
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - ONNX  (extension '.onnx')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file ../training/classification/models/cat_dog/resnet18.onnx.1.1.7103.GPU.FP16.engine
[TRT]    loading network plan from engine cache... ../training/classification/models/cat_dog/resnet18.onnx.1.1.7103.GPU.FP16.engine
[TRT]    device GPU, loaded ../training/classification/models/cat_dog/resnet18.onnx
[TRT]    Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.
[TRT]    Deserialize required 35057804 microseconds.
[TRT]    
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       29
[TRT]       -- maxBatchSize 1
[TRT]       -- workspace    0
[TRT]       -- deviceMemory 29827072
[TRT]       -- bindings     2
[TRT]       binding 0
                -- index   0
                -- name    'input_0'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  4
                -- dim #0  1 (SPATIAL)
                -- dim #1  3 (SPATIAL)
                -- dim #2  224 (SPATIAL)
                -- dim #3  224 (SPATIAL)
[TRT]       binding 1
                -- index   1
                -- name    'output_0'
                -- type    FP32
                -- in/out  OUTPUT
[... 61 lines cut ...]
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1600
  -- height:     900
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[image] loaded '../training/classification/data/cat_dog/test/cat/01.jpg'  (500x334, 3 channels)
class 0000 - 0.999999  (cat)
[OpenGL] glDisplay -- set the window size to 500x334
[OpenGL] creating 500x334 texture (GL_RGB8 format, 501000 bytes)
[cuda]   registered openGL texture for interop access (500x334, GL_RGB8, 501000 bytes)
[image] invalid extension format './training/classification/data/cat_dog/test/test_cat_result' saving image '../training/classification/data/cat_dog/test/test_cat_result'
[image] valid extensions are:  JPG/JPEG, PNG, TGA, BMP.
[image] imageWriter -- failed to save '../training/classification/data/cat_dog/test/test_cat_result'
Traceback (most recent call last):
  File "imagenet.py", line 79, in <module>
    output.Render(img)
Exception: jetson.utils -- videoOutput failed to render image**

closed 2023-04-27 by dusty-nv