dusty-nv/jetson-inference #857: git clone produces 2 pairs of files with identical names but one is capitalized -> could cause path collision on certain file systems

cpoptic, 2020-12-20:
Cloning repo throws warning
`git clone https://github.com/dusty-nv/jetson-inference.git`

> Cloning into 'jetson-inference'...
> remote: Enumerating objects: 16633, done.
> remote: Total 16633 (delta 0), reused 0 (delta 0), pack-reused 16633
> Receiving objects: 100% (16633/16633), 109.34 MiB | 1.41 MiB/s, done.
> Resolving deltas: 100% (13671/13671), done.
> Updating files: 100% (1617/1617), done.
>
> warning: the following paths have collided (e.g. case-sensitive paths
> on a case-insensitive filesystem) and only one from the same
> colliding group is in the working tree:
> 
>   'python/python/Jetson/__init__.py'
>   'python/python/jetson/__init__.py'
>   'python/python/Jetson/Inference/__init__.py'
>   'python/python/jetson/inference/__init__.py'

dusty-nv, 2020-12-21:
Are you cloning this onto a NTFS or FAT drive?    If so, please try partitioning it as ext3/ext4.

cpoptic, 2020-12-21:
Ah yes, I was cloning to an NTFS drive.  I'll redo on ext4 partition.  Thanks

closed 2020-12-21 by cpoptic