dusty-nv/jetson-inference #861: I'd like to train on a PC.

ingbeeedd, 2020-12-22:
Hi. After training on an external PC, I want to convert the model to onnx, so do I have to create an external pytorch transfer learning environment? I just need to bring the externally created model file and convert onnx format in jetson? right?

dusty-nv, 2020-12-22:
Hi @ingbeeedd , you should be able to convert the PyTorch model to ONNX on your PC, and just copy the ONNX model to your Jetson.

ingbeeedd, 2020-12-24:
What's more, is there no problem if I can convert to onnx using any pre-built model from the outside? Then I can deploy and use it right away on the board, right? I just wanted to ask you a few questions before I go through trial and error. The current labelling tool is made up of pascal voc.

closed 2023-04-27 by dusty-nv