dusty-nv/jetson-inference #862: The examples flash the image so the image can not be seen

JohnVorwald, 2020-12-22:
I would add some information to the example.  First, when the example code is first ran, it appears to train the network.  So, what is the input, output, and how long would be useful information.  Second, on the first, and subsequent runs, an image is flashed, but not visible, and then the classification is printed with a percentage.  Is it possible to display the image long enough to see it.  And how similar is the image to the training data?  And, what does the output mean?

dusty-nv, 2020-12-22:
The first time each model is run, it loaded with TensorRT (which takes some time to optimize the model the first run) - this is noted in the docs and in the terminal output.   It is not training the network or changing the outputs.   After the first run, the optimized model is cached to disk and re-loaded quicker on subsequent runs.

To see the output image, navigate to it in your file browser (i.e. `/path/to/your/jetson-inference/data/images/test/jellyfish.jpg`) and open it that way.   The OpenGL display window of the application is primarily intended to be used for video streams and cameras.  The window doesn't know it only is going to display one image, so it is only visible very briefly before the application shuts down.

JohnVorwald, 2020-12-26:
I believe dusty answered the question

closed 2020-12-26 by JohnVorwald