dusty-nv/jetson-inference #863: about keyboard events with "videoOutput"

lzcchl, 2020-12-25:
hello, I see that the class "glDisplay" and the class "videoOutput", Does "videoOutput" support keyboard events？can you show some guides or examples.

I see "gstCamera" not recommended to use, is that I should use "videoSource" + "glDisplay" instead of “videoSource” + "videoOutput" for keyboard events?

dusty-nv, 2020-12-28:
Hi @lzcchl , videoOutput interface doesn't support keyboard events (since only the GUI windows have those).   However you can cast the videoOutput interface to glDisplay and access the user events that way:

``` cpp
#include "glDisplay.h"
#include <X11/keysymdef.h>
...
if( output->IsType<glDisplay>() )
{
     glDisplay* display = (glDisplay*)output;	// safe to cast

     if( display->GetKey(XK_a) )
           // the 'a' key is pressed
}
```

lzcchl, 2020-12-29:
thanks your answer! I had use "glDisplay" class finish my work.

 but your suggest looks better, I will try it.

closed 2023-04-27 by dusty-nv