dusty-nv/jetson-inference #864: Equivalent gstreamer pipeline

McLovinfromHawaii, 2020-12-25:
Hello, I need an equivalent gstreamer pipeline to this command below
`video-viewer v4l2:///dev/video0 rtp://<remote-ip>:1234`
thanks in advance

dusty-nv, 2020-12-28:
Hi @McLovinfromHawaii , the GStreamer pipeline is constructed dynamically based on the input formats/codecs that your V4L2 camera uses (and which RTP codec is selected - by default H.264).   For example, if your V4L2 camera supports MJPEG, it will use mjpeg decoder in the pipeline (likewise if your camera uses H.264, ect).

The easiest way to get the pipeline for your setup would be to just run the video-viewer program, and it will print out the gstreamer pipeline it runs near the top of the terminal log.

McLovinfromHawaii, 2020-12-29:
Hi @dusty-nv  thanks for responding swiftly. Below is my output. I tried gstreamer pipeline generated but I cannot capture the stream from host device by using this pipeline on host. What am I doing wrong?
`gst-launch-1.0 -v udpsrc port=1234 \
 caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" ! \
 rtph264depay ! decodebin ! videoconvert ! autovideosink`

> [gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: VOLERS HD CAMERA
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"VOLERS\ HD\ CAMERA", v4l2.device.bus_info=(string)usb-3530000.xhci-1, v4l2.device.version=(uint)264588, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 12 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [6] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [7] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [8] image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [9] image/jpeg, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [10] image/jpeg, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [11] image/jpeg, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] gstCamera -- selected device profile:  codec=raw format=yuyv width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 ! video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720 ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      raw
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=4000000 ! video/x-h264 !  rtph264pay config-interval=1 ! udpsink host=192.168.8.111 port=1234 auto-multicast=true
[video]  created gstEncoder from rtp://192.168.8.111:1234
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: rtp://192.168.8.111:1234
     - protocol:  rtp
     - location:  192.168.8.111
     - port:      1234
  -- deviceType: ip
  -- ioType:     output
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  30.000000
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[... 102 lines cut ...]
video-viewer:  captured 24 frames (1280 x 720)
video-viewer:  captured 25 frames (1280 x 720)
video-viewer:  captured 26 frames (1280 x 720)
video-viewer:  captured 27 frames (1280 x 720)
[gstreamer] gstEncoder -- pipeline full, skipping frame (1382400 bytes)
video-viewer:  captured 28 frames (1280 x 720)
[gstreamer] gstEncoder -- pipeline full, skipping frame (1382400 bytes)
video-viewer:  captured 29 frames (1280 x 720)
[gstreamer] gstEncoder -- pipeline full, skipping frame (1382400 bytes)
video-viewer:  captured 30 frames (1280 x 720)
[gstreamer] gstEncoder -- pipeline full, skipping frame (1382400 bytes)
video-viewer:  captured 31 frames (1280 x 720)
[gstreamer] gstEncoder -- pipeline full, skipping frame (1382400 bytes)
video-viewer:  captured 32 frames (1280 x 720)
[gstreamer] gstEncoder -- pipeline full, skipping frame (1382400 bytes)
^Creceived SIGINT
video-viewer:  captured 33 frames (1280 x 720)
[gstreamer] gstEncoder -- pipeline full, skipping frame (1382400 bytes)
video-viewer:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped
[gstreamer] gstEncoder -- shutting down pipeline, sending EOS
[gstreamer] gstEncoder -- transitioning pipeline to GST_STATE_NULL
[gstreamer] gstEncoder -- pipeline stopped
video-viewer:  shutdown complete

dusty-nv, 2020-12-29:
Do you mean you want the GStreamer pipeline to view the RTP stream?

If so, please see here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#viewing-rtp-remotely

closed 2021-01-01 by McLovinfromHawaii

irshadkn, 2021-07-19:
import jetson.utils

input = jetson.utils.videoSource("rtsp://@192.168.1.61:554/Streaming/channels/102")
output = jetson.utils.videoOutput("")

while output.IsStreaming():
	image = input.Capture()
	output.Render(image)
	output.SetStatus("Video Viewer | {:d}x{:d} | {:.1f} FPS".format(image.width, image.height, output.GetFrameRate()))

This code is working for me. But there is a lag(latency). I know there is an option with gstreamer pipeline to reduce latency. How to do with this code? Otherwise, Is there any option to pass gstreamer pipeline in this code?