dusty-nv/jetson-inference #865: Live Camera Stream Upside down

Sourish07, 2020-12-26:
https://www.youtube.com/watch?v=bcM5AQSAzUY&ab_channel=NVIDIADeveloper

I followed everything in that video, yet my video stream is upside down. I am using the command line argument --input-flip=rotate-180 with my py script but that does not solve the issue. Any help would be appreciated.

dusty-nv, 2020-12-28:
Hi @Sourish07, which script are you running?   Can you try running the `video-viewer` utility first to see if the `--input-flip=rotate-180` argument takes effect there?    

Note that this argument is only applicable to MIPI CSI cameras, and not V4L2 USB webcams.

Sourish07, 2020-12-29:
Thank you for the response. I am running a py script on my desktop titled "my-detection.py". The contents of the script are the exact 10 lines from the YouTube video. Also, I am indeed using a MIPI CSI camera. 

After running the following command: python3 video-viewer.py /dev/video0 --input-flip=rotate-180, with the working directory being: /jetson-inference/build/aarch64/bin, I get the following output (I don't know how to unpause the gstreamer):

[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0

(python3:9243): GStreamer-CRITICAL **: 19:40:35.872: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed

(python3:9243): GStreamer-CRITICAL **: 19:40:35.872: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed
[gstreamer] gstCamera -- didn't discover any v4l2 devices
[gstreamer] gstCamera -- device discovery failed, but /dev/video0 exists
[gstreamer]              support for compressed formats is disabled
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      unknown
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: rotate-180
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  2560x1440
[OpenGL] glDisplay -- X window resolution:    2560x1440
[OpenGL] glDisplay -- display device initialized (2560x1440)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      2560
  -- height:     1440
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstCamera -- end of stream (EOS)
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> v4l2src0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> v4l2src0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer message stream-start ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> v4l2src0
[gstreamer] gstreamer v4l2src0 ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink

dusty-nv, 2020-12-29:
> Also, I am indeed using a MIPI CSI camera.

OK, can you try running this command instead?

``` bash
video-viewer --input-flip=rotate-180 csi://0
```

Sourish07, 2020-12-30:
Yes, that does fix the issue for the video-viewer script. How do I transfer that fix to my-detection.py? I tried adding the command line argument when I ran the script, but it still resulted in an upside-down video feed.

dusty-nv, 2020-12-30:
Try using detectnet.py with the same command line argument instead.  It is a very similar script to my-detection.py but supports more advanced options.

closed 2023-04-27 by dusty-nv