dusty-nv/jetson-inference #866: gstDecoder -- failed to discover stream info

SergeyMalashenko, 2020-12-26:
Good day! I am trying to debug my code. So I see the following random message.

[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for ./e36-test-29294ba2ae_1602173004_587.mp4
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for file:///home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/build_release/./e36-test-29294ba2ae_1602173004_587.mp4

Why can it be?

SergeyMalashenko, 2020-12-27:
I investigated the problem under debugger. So I have the following stack
```
#0  0x00007ffff7bc51fd in __lll_lock_wait () at ../sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:135
#1  0x00007ffff7bbe0f4 in __GI___pthread_mutex_lock (mutex=0x7fff88010c10) at ../nptl/pthread_mutex_lock.c:115
#2  0x00007fffe73cf86f in post_activate (new_mode=GST_PAD_MODE_NONE, pad=0x7fff84024850) at gstpad.c:1053
#3  0x00007fffe73cf86f in activate_mode_internal (pad=pad@entry=0x7fff84024850, parent=parent@entry=0x7fff88010850, mode=mode@entry=GST_PAD_MODE_PUSH, active=active@entry=0) at gstpad.c:1231
#4  0x00007fffe73d0325 in gst_pad_set_active (pad=pad@entry=0x7fff84024850, active=0) at gstpad.c:1122
#5  0x00007fffe73adf0d in activate_pads (vpad=<optimized out>, ret=0x7fffffffd290, active=0x7fffffffd2ec) at gstelement.c:3040
#6  0x00007fffe73c0874 in gst_iterator_fold (it=it@entry=0x555555e3b8a0, func=func@entry=0x7fffe73adef0 <activate_pads>, ret=ret@entry=0x7fffffffd290, user_data=user_data@entry=0x7fffffffd2ec)
    at gstiterator.c:617
#7  0x00007fffe73aea16 in iterator_activate_fold_with_resync (iter=iter@entry=0x555555e3b8a0, user_data=user_data@entry=0x7fffffffd2ec, func=0x7fffe73adef0 <activate_pads>) at gstelement.c:3064
#8  0x00007fffe73b095e in gst_element_pads_activate (element=element@entry=0x7fff88010850, active=<optimized out>, active@entry=0) at gstelement.c:3108
#9  0x00007fffe73b0c06 in gst_element_change_state_func (element=0x7fff88010850, transition=<optimized out>) at gstelement.c:3174
#10 0x00007fffbd933365 in  () at /usr/lib/x86_64-linux-gnu/libgstvideo-1.0.so.0
#11 0x00007fff91d89b53 in gst_v4l2_video_dec_change_state () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/deepstream/libgstnvvideo4linux2.so
#12 0x00007fffe73b2d5e in gst_element_change_state (element=element@entry=0x7fff88010850, transition=transition@entry=GST_STATE_CHANGE_PAUSED_TO_READY) at gstelement.c:2952
#13 0x00007fffe73b3499 in gst_element_set_state_func (element=0x7fff88010850, state=GST_STATE_READY) at gstelement.c:2906
#14 0x00007fffe7390a02 in gst_bin_element_set_state (next=GST_STATE_READY, current=GST_STATE_PAUSED, start_time=0, base_time=0, element=0x7fff88010850, bin=0x555555f76380) at gstbin.c:2604
#15 0x00007fffe7390a02 in gst_bin_change_state_func (element=0x555555f76380, transition=GST_STATE_CHANGE_PAUSED_TO_READY) at gstbin.c:2946
#16 0x00007fff94752029 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so
#17 0x00007fffe73b2d5e in gst_element_change_state (element=element@entry=0x555555f76380, transition=transition@entry=GST_STATE_CHANGE_PAUSED_TO_READY) at gstelement.c:2952
#18 0x00007fffe73b3499 in gst_element_set_state_func (element=0x555555f76380, state=GST_STATE_READY) at gstelement.c:2906
#19 0x00007fffe7390a02 in gst_bin_element_set_state (next=GST_STATE_READY, current=GST_STATE_PAUSED, start_time=0, base_time=0, element=0x555555f76380, bin=0x555555f70090) at gstbin.c:2604
#20 0x00007fffe7390a02 in gst_bin_change_state_func (element=0x555555f70090, transition=GST_STATE_CHANGE_PAUSED_TO_READY) at gstbin.c:2946
#21 0x00007fff9476569a in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so
#22 0x00007fffe73b2d5e in gst_element_change_state (element=element@entry=0x555555f70090, transition=transition@entry=GST_STATE_CHANGE_PAUSED_TO_READY) at gstelement.c:2952
#23 0x00007fffe73b3499 in gst_element_set_state_func (element=0x555555f70090, state=GST_STATE_READY) at gstelement.c:2906
#24 0x00007fffe7390a02 in gst_bin_element_set_state (next=GST_STATE_READY, current=GST_STATE_PAUSED, start_time=0, base_time=0, element=0x555555f70090, bin=0x555555f68160) at gstbin.c:2604
#25 0x00007fffe7390a02 in gst_bin_change_state_func (element=0x555555f68160, transition=GST_STATE_CHANGE_PAUSED_TO_READY) at gstbin.c:2946
#26 0x00007fffe73b2d5e in gst_element_change_state (element=element@entry=0x555555f68160, transition=transition@entry=GST_STATE_CHANGE_PAUSED_TO_READY) at gstelement.c:2952
#27 0x00007fffe73b3499 in gst_element_set_state_func (element=0x555555f68160, state=GST_STATE_READY) at gstelement.c:2906
#28 0x00007fffe6f2b5d9 in  () at /usr/lib/x86_64-linux-gnu/libgstpbutils-1.0.so.0
#29 0x00007fffe6f2fdfb in gst_discoverer_discover_uri () at /usr/lib/x86_64-linux-gnu/libgstpbutils-1.0.so.0
#30 0x000055555559fa4d in gstDecoder::discover() (this=0x555555a22a80) at /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/external/codec/gstDecoder.cpp:332
#31 0x000055555559f446 in gstDecoder::init() (this=0x555555a22a80) at /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/external/codec/gstDecoder.cpp:198
#32 0x000055555559f195 in gstDecoder::Create(videoOptions const&) (options=...) at /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/external/codec/gstDecoder.cpp:144
#33 0x00005555555e84ec in videoSource::Create(videoOptions const&) (options=...) at /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/external/video/videoSource.cpp:54
#34 0x00005555555e89c7 in videoSource::Create(commandLine const&, int) (cmdLine=..., positionArg=0)
    at /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/external/video/videoSource.cpp:129
```

dusty-nv, 2020-12-28:
The initial error message states need to set the --input-codec command line argument because GStreamer failed to discover it from the video file.  For example if the videos codec is H264, set --input-codec=h264

SergeyMalashenko, 2020-12-28:
Thanks. For me now interesting the second problem. I have two videos on the first one now all is OK. But the second one freezes my application

Uploading test_5.mp4…

gcordova19, 2021-06-29:
Hi!
In my case I put :
ros2 launch ros_deep_learning video_viewer.ros2.launch 
 input:="/home/9f1de61c-3123-4e27-bc89-3a317bb6973a.mp4" output:=display://0  --input-codec=h264 

and I got
ros2: error: unrecognized arguments: --input-codec=h264 

it's wrong? I have tried in different positions and it does not work

dusty-nv, 2021-06-29:
Hi @gcordova19, I think you need to launch it like `input_codec:=h264` - see below

``` bash
ros2 launch ros_deep_learning video_viewer.ros2.launch input:="/home/9f1de61c-3123-4e27-bc89-3a317bb6973a.mp4" output:=display://0 input_codec:=h264
```

gcordova19, 2021-07-01:
Hi @dusty-nv ,
I have launched it as you told me but it does not work

```
~/ros2_example_ws$ ros2 launch ros_deep_learning video_viwer.ros2.launch input:="/home/gcordova/9f1de61c-3123-4e27-bc89-3a317bb6973a.mp4" output:=display://0 input_code:=h264
file 'video_viwer.ros2.launch' was not found in the share directory of package 'ros_deep_learning' which is at '/home/gcordova/ros2_example_ws/install/ros_deep_learning/share/ros_deep_learning'
```

I have checked the forum thread [forum](https://github.com/dusty-nv/ros_deep_learning/issues/45)

I add AMENT_PREFIX_PATH= /home/ros2_example_ws/install/ros_deep_learning:/opt/ros/foxy on ~/.bashrc
Then on ~/ros2_example_ws run
ros2 launch ros_deep_learning video_viwer.ros2.launch input:="/home/gcordova/9f1de61c-3123-4e27-bc89-3a317bb6973a.mp4" output:=display://0 input_code:=h264

and I got
file 'video_viwer.ros2.launch' was not found in the share directory of package 'ros_deep_learning' which is at '/home/gcordova/ros2_example_ws/install/ros_deep_learning/share/ros_deep_learning'

What I can do?

closed 2023-04-27 by dusty-nv