dusty-nv/jetson-inference #869: My dig is pissed

tompizatscania, 2020-12-28:
[image]
Well, somthing does not go right when i train the cat & dog modell, geting really poor results. Have tried to resolve this a couple of days now. When i mount the pretrained ( 100 epochs ) it works fine but when i do one epoch like in the video, the model classifies 97 % of the dogs as cats.

dusty-nv, 2020-12-28:
Hi @tompizatscania , what accuracy does your model train to in PyTorch?

Also, can you delete the `*.engine` file from your model's folder and try running the imagenet program again?

tompizatscania, 2020-12-29:
Test: [230/250]  Time  0.143 ( 0.137)  Loss 3.8741e-01 (1.1304e+00)  Acc@1 100.00 ( 45.89)  Acc@5 100.00 (100.00)
Test: [240/250]  Time  0.128 ( 0.137)  Loss 3.7522e-01 (1.1008e+00)  Acc@1 100.00 ( 48.13)  Acc@5 100.00 (100.00)
 * Acc@1 50.000 Acc@5 100.000
saved best model to:  models/dog_cat/model_best.pth.tar
root@thomas-desktop:/jetson-inference/python/training/classification# python3 onnx_export.py --model-dir=models/dog_cat
Namespace(input='model_best.pth.tar', model_dir='models/dog_cat', no_softmax=False, output='')
running on device cuda:0
loading checkpoint:  models/dog_cat/model_best.pth.tar
using model:  resnet18

tompizatscania, 2020-12-29:
I tried to delete the *.engine file but same same...

dusty-nv, 2020-12-29:
It appears the accuracy has only trained to 50%, whereas I believe the one I trained was ~80%.   How many epochs have you trained it for?

closed 2023-04-27 by dusty-nv