dusty-nv/jetson-inference #870: Retrained ResNet152 ONNX model - "Dynamic Shape....No Optimization.."

DJT777, 2020-12-28:
Hey Dusty,

I have retrained the ResNet152 as a bird species classifier using a tutorial I found [elsewhere](https://github.com/ecm200/caltech_birds). The examples folder in that repository has the final notebook which displays how the library is exported as ONNX. I have looked through the code, and I have found where it has specified optimization. I am at a loss for how to solve this issue. I've been searching around on how to just build my own program in DeepStream, but I have a steep learning curve. If you could help me get this .ONNX model properly loaded into your repo it would save me some time in getting the first version of my system that I am building up and running.

Please have a look at the final notebook in the example notebooks folder and/or help me figure out how to get my network properly optimized so I can deploy.

Thanks

dusty-nv, 2020-12-28:
Hi @DJT777 , is the ONNX model exported with dynamic shapes?   Does it set the `dynamic_axes` argument to `torch.onnx.export()`?

If so, my code doesn't use dynamic shapes and is exported with a static input size, like in this script:

https://github.com/dusty-nv/pytorch-classification/blob/15c7d51cd6a1271e5ee6959c3f91fb95b6b54ff0/onnx_export.py

Also, can the trtexec program load your ONNX model?   It is found under `/usr/src/tensorrt/bin`.

DJT777, 2020-12-31:
Dusty, I think that this issue is moot because I have been trying to use a classification model for a detection purpose. 

I was able to cover the onnx to an engine and deploy it to deepstream with a custom config. However I discovered that I needed to parse bounding box outputs, and realized my classification model was not what I needed.

I am learning a lot about this kind of c.s.. My goal now is to train a model such as Faster-RCNN or DetectNet that nVidia has already configured for Deepstream.

I tried to train with your DIGITS tutorial, but I don't have a local GPU able to handle it right now. (Jetson is not doing training well) So I've opted to retrain  DetectNet/F-RCNN/SDD/YOLO in hopes Deepstream will not need too much modification.

Hopefully with a trained Detection model that is compatible with Deepstream I will be able to implement a prototype of my bird detector/classifier.

Question before you close: Are existing models, such as the ones I have mentioned, able to perform detection of class of something in similar shape but different colors, i.e. birds. I know detectors seem to distinguish between easily discernable objects like signs or cars, but can they distinguish and classify something like a bird species? What about deploying a detector, have it identify the bounding box, and then pass the image within the bounding box to a classifier? Is that feasible? Question before you close.

Feel free to close this on the grounds that the problem has been solved.

closed 2021-01-08 by DJT777