dusty-nv/jetson-inference #872: How can I use mobileenet_v2 as classification about transfer learning?

ingbeeedd, 2020-12-29:
Hi

After setting up the PC environment, I am going to move the code to the PC and run it successfully. Pytorch provides mobileenet_v2 as a [pre-trained model](https://pytorch.org/docs/stable/torchvision/models.html), but an error occurs in this code. How can I solve this error? I'm wondering if you're not letting me do it, because I think the actual code makes an exception.

[image]

dusty-nv, 2020-12-29:
Transfer learning often requires re-shaping of the network's output layer (to go from the typical 1000-class pre-trained models to the N-classes that you are training on).   The re-shaping operation is custom to each network type, and those are defined here in reshape.py:

https://github.com/dusty-nv/pytorch-classification/blob/master/reshape.py

So you would need to add the support for mobilenet architecture to that code.  You basically look up the original code that created the model in PyTorch, and then replace the output layer with one with the desired number of outputs.

ingbeeedd, 2020-12-31:
Thank you. I understand that. I try to convert code for mobileNet_v2

closed 2023-04-27 by dusty-nv