dusty-nv/jetson-inference #874: Memory issues after I loaded my models.

SergeyMalashenko, 2020-12-29:
Good day!

I see the following messages after I loaded all required models. Please help me understand the problem.
```
RingBuffer -- allocated 4 buffers (3110415 bytes each, 12441660 bytes total)
RingBuffer -- allocated 4 buffers (24883200 bytes each, 99532800 bytes total)
RingBuffer -- allocated 4 buffers (3110431 bytes each, 12441724 bytes total)
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/tensorRT_LicensePlatesVideoProcessing_main.cpp:105
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/external/cuda/cudaYUV-YV12.cu:119
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/external/cuda/cudaColorspace.cpp:55
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/smalashenko/My_projects/C++/licenseplatesdetectionrecognition_tensorrt/external/codec/gstDecoder.cpp:896
[gstreamer] gstDecoder::Capture() -- unsupported image format (rgb32f)
[gstreamer]                          supported formats are:
[gstreamer]                              * rgb8
[gstreamer]                              * rgba8
[gstreamer]                              * rgb32f
[gstreamer]                              * rgba32f
```

dusty-nv, 2021-01-05:
The memory is setup for zero-copy for Jetson (i.e. CUDA mapped memory), and while it should still work on PC, I haven't tried it.  

What is the code that throws the first memory error?

SergeyMalashenko, 2021-01-12:
I hope that the problem on my side. I fixed several issues and continued to test my project,

closed 2021-01-12 by SergeyMalashenko