dusty-nv/jetson-inference #875: Image Crop isn't working

tkislan, 2020-12-31:
Even when using example implementation in `build/aarch64/bin/cuda-examples.py`, it produces invalid images when cropping
Not sure what is this related to, because when tracked back to cuda implementation, it seems, that is just copies the underlying memory, which is just a 1 dimensional array .. so no idea why the images gets messed up like this
```
output[out_y * outWidth + out_x] = input[in_y * inWidth + in_x];
```

[image]

marcelosantoro, 2021-01-02:
Same problem here :(

dusty-nv, 2021-04-26:
Hi all, please see this post for similar image and make sure that your ROI is specified correctly:

https://forums.developer.nvidia.com/t/cudacrop-mangling-my-image/166467/7?u=dusty_nv

Can you also check that this sample code works correctly for you:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#cropping

closed 2023-04-27 by dusty-nv