dusty-nv/jetson-inference #876: How should Do inference  using own resnet model?

bundelesneha05, 2021-01-02:
I have trained the resnet18 model on the custom dataset. I would like to include that in the network available. what are the steps to be followed?

dusty-nv, 2021-01-04:
Hi @bundelesneha05 - was it trained with PyTorch and has it been converted to ONNX?

If so, you can run it like this by specifying your custom model arguments on the command line:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md#processing-images-with-tensorrt

bundelesneha05, 2021-01-05:
Thanks for the reply. It has been converted to the ONNX. The above link worked.

bundelesneha05, 2021-01-05:
I have also trained the ssd_fpn_resnet_18 model using TensorFlow(1.15.1) on a custom dataset. The model has converted to the ONNX format. Not getting how to do the inference for this model using detectnet.py.

dusty-nv, 2021-01-05:
Supporting different model architectures requires that the pre/post-processing code be added to jetson-inference/c/detectNet.cpp

The code supporting the ONNX models for detectNet.cpp/.py are currently setup for the SSD-Mobilenet models trained with PyTorch like here:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md

closed 2023-04-27 by dusty-nv