dusty-nv/jetson-inference #877: Training with existing PascalVOC Format Data

naimulhq, 2021-01-03:
I have been working on a project which involves coming up with an object detection model. At first, I was using ImageAI to train and run an object detection model but it ended up running out of memory and running to slow on the Jetson Nano. Now, I have switched to this package and I am trying to figure out how to train with already existing data? I already have all the images and annotations for each of the images but am unsure of how to set up the directory and its structures.

neilyoung, 2021-01-03:
@naimulhq Did you figure it out?

naimulhq, 2021-01-03:
@neilyoung Yes. Another existing issue helped out alot when trying to figure it out. #789

neilyoung, 2021-01-03:
@naimulhq Thanks for sharing!

closed 2021-01-09 by naimulhq