dusty-nv/jetson-inference #880: How to solve the problem of multiple detection for single object in a single frame?

Asmita2695, 2021-01-04:
Hi,
I have trained a single class SSD mobilenet object detector using images from OpenImage Dataset. When used trained model with live stream it is detecting the same object multiple times with the same class in a single frame. How can I solve this?
Thank you.

dusty-nv, 2021-01-05:
Can you try decreasing the threshold parameter in the code here:

https://github.com/dusty-nv/jetson-inference/blob/0fd4a76101be5c9af4723a9e8376018dfa462be4/c/detectNet.h#L477

Then re-run `make` and `sudo make install`
This threshold parameter controls how much overlap before bounding boxes are clustered.

Asmita2695, 2021-01-06:
I will try this. Thank you.

lpeleato2019, 2022-11-05:
> @rajeshroy402 the same one as my comment from above. Try decreasing the `threshold=0.75f` argument and rebuiding/reinstall.

Hello @dusty-nv. I tried lowering the threshold (variable: DETECTNET_DEFAULT_CLUSTERING_THRESHOLD) in the detectNet.h file but I'm getting the same result. I'm attaching both pictures using the default 0.75f and 0.25f threshold.

Thank you.

[image]

[image]

dusty-nv, 2022-11-06:
Hi @lpeleato2019, did you re-run `make && sudo make install` after making the change?

lpeleato2019, 2022-11-06:
> Hi @lpeleato2019, did you re-run `make && sudo make install` after making the change?

Hello Dusty, I did

dusty-nv, 2022-11-06:
Heya, what's the model?   Is it a ONNX, TensorFlow UFF, or caffemodel?

Does using the `--clustering=0.25` command-line argument to detectnet/detectnet.py work for you?

lpeleato2019, 2022-11-06:
> 

It's a ONNX model. I tried using --clustering=0.25 but it did not work.

This is the command I'm running: 

detectnet --model=models/dasani/ssd-mobilenet.onnx --clustering=0.25 --labels=models/dasani/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "/jetson-inference/data/images/bottle/bottle_*.JPEG" /jetson-inference/data/images/test2/bottle_test_%i.jpg

lpeleato2019, 2022-11-06:
> Heya, what's the model? Is it a ONNX, TensorFlow UFF, or caffemodel?
> 
> Does using the `--clustering=0.25` command-line argument to detectnet/detectnet.py work for you?

It's a ONNX model. I tried using --clustering=0.25 but it did not work.

This is the command I'm running: 

detectnet --model=models/dasani/ssd-mobilenet.onnx --clustering=0.25 --labels=models/dasani/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "/jetson-inference/data/images/bottle/bottle_*.JPEG" /jetson-inference/data/images/test2/bottle_test_%i.jpg

dusty-nv, 2022-11-07:
Would you mind sending me the model somehow so I can reproduce it?  Thanks.

luispeleato, 2022-11-07:
> Would you mind sending me the model somehow so I can reproduce it? Thanks.

Not at all. I just sent it to your email at dustinf@nvidia.com. with the subject "Problem of multiple detection for single object"

dusty-nv, 2022-11-08:
OK, I found another bottles image to ran and confirmed that `--clustering=0.2` does indeed have an effect.

What I noticed from your command, is that it looks like you are using the container – and depending on which version of JetPack-L4T you are using, I may not have rebuilt the jetson-inference container to support those arguments.

So I would recommend trying to build the project from source and try running it outside of container to see if that helps:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md

If you decide to rebuild the container based on the latest code with docker/build.sh, *make sure* that you backup your model and dataset first, because that docker/build.sh script will data under jetson-inference/python/training/detection/ssd (so that it doesn’t get copied into the container)

Quasimondo, 2022-11-24:
Not sure if it helps in your case, but in the Python version I could make it work by calling SetClusteringThreshold after the model has been instantiated, e.g.:

`net = detectNet(args.network, sys.argv, args.threshold)
net.SetClusteringThreshold(0.25)`

luispeleato, 2022-11-28:
Hello @dusty-nv. As you suggested, I built the project from the source and executed it outside of the container. That worked really well! Thank you for your help.

Now, instead of passing image files, I am trying to feed it live images from a camera. I'm running the command indicating that my input is at /dev/video0 instead of the image files. However, when I run it, I get the following error:
```
[gstreamer] gstDecoder -- failed to retrieve next image buffer
detectnet: failed to capture video frame
```

I saw [this suggestion that you made in the Nvidia dev forums](https://forums.developer.nvidia.com/t/gstreamer-gstdecoder-failed-to-retrieve-next-image-buffer/195168) but I am having problems running `cmake -DENABLE_NVMM=OFF ../`:
```
CMake Geberate step failed. Build files cannot be regenerated correctly.
```

Any ideas what I could be doing wrong? Thank you.

closed 2023-04-27 by dusty-nv